Xilinx/Vitis-Tutorials #3: How to add the example platforms to Vitis?

niclasb, 2019-11-25:
How do you add the required platforms to Vitis to be able to run the examples?

ERROR: [v++ 60-1258] No valid platform was found that matches 'zcu102_base'. Please make sure that the platform is specified correctly, and the platform has the right version number. The platform repo paths are:
	/xilinx/Vitis/2019.2/platforms
The valid platforms found from the above repo paths are:
	/xilinx/Vitis/2019.2/platforms/xcvc1902_fixed/xcvc1902_fixed.xpfm

ERROR: [v++ 60-587] Failed to add a platform: specified platform zcu102_base is not found or is not valid
ERROR: [v++ 60-592] Failed to finish compilation
Makefile:95: recipe for target '_x.sw_emu.zcu102_base/vadd.xo' failed
make: *** [_x.sw_emu.zcu102_base/vadd.xo] Error 1

niclasb, 2019-12-03:
I think I understand what went wrong now. The DEVICE argument should be a full path instead of just a devicename. Setting the argument like this solved the above problem: DEVICE=/xilinx/vitis/zcu102_base/zcu102_base.xpfm

gkrislara, 2019-12-17:
I am new to Vitis and I have a similar issue 
I downloaded alveo U200 target platform in ubuntu 16.04 but i couldnt find .xpfm file or directory containing it. I dont know where I am going wrong 
and if at all I need to change the DEVICE argument , where will it be?

Ssicayoon, 2019-12-21:
> I think I understand what went wrong now. The DEVICE argument should be a full path instead of just a devicename. Setting the argument like this solved the above problem: DEVICE=/xilinx/vitis/zcu102_base/zcu102_base.xpfm

Are you able to do sw/hw_emulation of zcu102 on host (x86_64)?
I can only test the actual performance by inserting SD card to zcu102, but no emulation on host.

closed 2021-02-20 by rwarmstr