Xilinx/Vitis-Tutorials #9: unable to run hw_emu

dharawolvpatel, 2020-05-28:
Building these tutorials works fine when using sw_emu. But I'm unable to build these applications using hw_emu. I always get the following errors: 
[image]

Ssicayoon, 2020-05-28:
I note that your run directory is not `makefile`.
[Run Hardware Emulation](https://github.com/Xilinx/Vitis-Tutorials/blob/master/docs/bloom/baseline_fpga.md#run-hardware-emulation) gives the command to run hw_emu.
Try it and see whether it works.

dharawolvpatel, 2020-05-28:
the directory I'm running "make" is the following: 
/home/dharap/xilinx/Vitis-Tutorials/Vitis-Tutorials/docs/bloom/design/makefile

Ssicayoon, 2020-05-28:
@dharawolvpatel 
Have you installed the platform `xilinx_u200_xdma_201830_2` after your installation of Vitis?

dharawolvpatel, 2020-05-28:
yes, its there! sw_emu works fine with it.

Ssicayoon, 2020-05-29:
@dharawolvpatel 
I have tried the `bloom` tutorial and no error occurs.
Notice that for [Kernel Code Modifications](https://github.com/Ssicayoon/Vitis-Tutorials/blob/master/docs/bloom/baseline_fpga.md#kernel-code-modifications), I copy `computer_score_fpga.cpp` and `compute_score_host.cpp` in `/design/refernce_files/baseline_fpga` to `/design/src/baseline_fpga`.
Have a try.

ryanvergel, 2020-06-01:
Thank you for your input. I will create an internal ticket so that the Author can review this issue.

closed 2021-02-24 by rwarmstr