Xilinx/Vitis-Tutorials #12: "dlopen <path>/dltmp is failed" error message

jrjbertram, 2020-07-14:
Hi,

I'm running into an issue trying to run software emulation mode for the zcu102 target.  The error message is:
`
ERROR: dlopen of /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/.run/19717/sw_emu/device0/binary_0/dltmp is failed. Please check undefined symbols in the kernel 
`

Not quite sure what I've done wrong.  Have been following the instructions in this repo with the exception of adapting them for the zcu102 platform.  Any ideas?

Thanks,
- J.

In my ~/.bashrc:
`
source /tools/Xilinx/Vitis/2020.1/settings64.sh
source /opt/xilinx/xrt/setup.sh
export PLATFORM_REPO_PATHS=/opt/xilinx/platforms/

export LIBRARY_PATH=/usr/lib/x86_64-linux-gnu
`

My design.cfg:
`
(base) adclab@solo:~/Vitis-Tutorials/docs/Pathway3/reference-files/run$ cat design.cfg 
#platform=xilinx_u200_xdma_201830_2
platform=xilinx_zcu102_base_202010_1
debug=1

[connectivity]
nk=mmult:1:mmult_1
`

Modified PLATFORM variable in run/Makefile:
`
 21 TARGET := sw_emu
 22 #PLATFORM := xilinx_u200_xdma_201830_2
 23 PLATFORM := xilinx_zcu102_base_202010_1
 24 HOST_EXE := host
 25 XO := mmult.$(TARGET).$(PLATFORM).xo
 26 XCLBIN := mmult.$(TARGET).$(PLATFORM).xclbin
 27 
`

Platforms I have installed:
`
(base) adclab@solo:~/Vitis-Tutorials/docs/Pathway3/reference-files/run$ ls $PLATFORM_REPO_PATHS 
xilinx_zcu102_base_202010_1
`

`
(base) adclab@solo:~/Vitis-Tutorials/docs/Pathway3/reference-files/run$ make run TARGET=sw_emu
v++ -t sw_emu --config design.cfg -c -k mmult -I'../src' -o'mmult.sw_emu.xilinx_zcu102_base_202010_1.xo' '../src/mmult.cpp'
Option Map File Used: '/tools/Xilinx/Vitis/2020.1/data/vitis/vpp/optMap.xml'

****** v++ v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/reports/mmult.sw_emu.xilinx_zcu102_base_202010_1
	Log files: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/logs/mmult.sw_emu.xilinx_zcu102_base_202010_1
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:35059
INFO: [v++ 60-1548] Creating build summary session with primary output /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/mmult.sw_emu.xilinx_zcu102_base_202010_1.xo.compile_summary, at Tue Jul 14 13:18:57 2020
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Tue Jul 14 13:18:57 2020
Running Rule Check Server on port:45237
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/reports/mmult.sw_emu.xilinx_zcu102_base_202010_1/v++_compile_mmult.sw_emu.xilinx_zcu102_base_202010_1_guidance.html', at Tue Jul 14 13:18:58 2020
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_zcu102_base_202010_1/xilinx_zcu102_base_202010_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_zcu102_base_202010_1/hw/xilinx_zcu102_base_202010_1.xsa'
INFO: [v++ 60-585] Compiling for software emulation target
INFO: [v++ 60-423]   Target device: xilinx_zcu102_base_202010_1
INFO: [v++ 60-242] Creating kernel: 'mmult'
INFO: [v++ 60-594] Finished kernel compilation
INFO: [v++ 60-586] Created mmult.sw_emu.xilinx_zcu102_base_202010_1.xo
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
    vitis_analyzer /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/mmult.sw_emu.xilinx_zcu102_base_202010_1.xo.compile_summary 
INFO: [v++ 60-791] Total elapsed time: 0h 0m 16s
INFO: [v++ 60-1653] Closing dispatch client.
v++ -t sw_emu --config design.cfg -l -o'mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin' mmult.sw_emu.xilinx_zcu102_base_202010_1.xo
Option Map File Used: '/tools/Xilinx/Vitis/2020.1/data/vitis/vpp/optMap.xml'

****** v++ v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ link can be found at:
	Reports: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/reports/link
	Log files: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/logs/link
INFO: [v++ 60-1657] Initializing dispatch client.
Running Dispatch Server on port:43803
INFO: [v++ 60-1548] Creating build summary session with primary output /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin.link_summary, at Tue Jul 14 13:19:15 2020
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Tue Jul 14 13:19:15 2020
Running Rule Check Server on port:37647
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/reports/link/v++_link_mmult.sw_emu.xilinx_zcu102_base_202010_1_guidance.html', at Tue Jul 14 13:19:16 2020
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_zcu102_base_202010_1/xilinx_zcu102_base_202010_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_zcu102_base_202010_1/hw/xilinx_zcu102_base_202010_1.xsa'
INFO: [v++ 60-629] Linking for software emulation target
INFO: [v++ 60-423]   Target device: xilinx_zcu102_base_202010_1
INFO: [v++ 60-645] kernel flags are '-g -I /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/src -g'
INFO: [v++ 60-586] Created mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin
INFO: [v++ 60-1307] Run completed. Additional information can be found in:
	Guidance: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/reports/link/v++_link_mmult.sw_emu.xilinx_zcu102_base_202010_1_guidance.html
	Steps Log File: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/_x/logs/link/link.steps.log

INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
    vitis_analyzer /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin.link_summary 
INFO: [v++ 60-791] Total elapsed time: 0h 0m 13s
INFO: [v++ 60-1653] Closing dispatch client.
g++ -I/opt/xilinx/xrt/include/ -I/tools/Xilinx/Vivado/2020.1/include/ -Wall -O0 -g -std=c++11 -L/opt/xilinx/xrt/lib/ -lpthread -lrt -lstdc++ -o 'host' '../src/host.cpp' -lOpenCL
emconfigutil --platform xilinx_zcu102_base_202010_1

****** configutil v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [ConfigUtil 60-895]   Target platform: /opt/xilinx/platforms/xilinx_zcu102_base_202010_1/xilinx_zcu102_base_202010_1.xpfm
INFO: [ConfigUtil 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_zcu102_base_202010_1/hw/xilinx_zcu102_base_202010_1.xsa'
emulation configuration file `emconfig.json` is created in current working directory 
XCL_EMULATION_MODE=sw_emu ./host mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin
Found Platform
Platform Name: Xilinx
INFO: Reading mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin
Loading: 'mmult.sw_emu.xilinx_zcu102_base_202010_1.xclbin'
ERROR: dlopen of /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/.run/19717/sw_emu/device0/binary_0/dltmp is failed. Please check undefined symbols in the kernel 
XRT build version: 2.6.655
Build hash: 2d6bfe4ce91051d4e5b499d38fc493586dd4859a
Build date: 2020-05-22 12:05:03
Git branch: 2020.1
PID: 19717
UID: 1000
[Tue Jul 14 13:19:21 2020]
HOST: solo
EXE: /home/adclab/Vitis-Tutorials/docs/Pathway3/reference-files/run/host
[XRT] ERROR: Failed to load xclbin.
../src/host.cpp:200 Error calling cl::Program program(context, devices, bins, NULL, &err), error code is: -44
Makefile:59: recipe for target 'run' failed
make: *** [run] Error 1
`

rwarmstr, 2021-02-20:
@jrjbertram Sincere apologies for the unreasonable delay responding on this issue. With the amount of passed time I'm going to close this out, but if you still have similar issues please do let us know.

closed 2021-02-20 by rwarmstr

SisenZhang, 2022-11-18:
您好，have you solved 头和problem? I have suffered the same now.