Xilinx/Vitis-Tutorials #16: failed software emulation: ../../src/host.cpp:33:10: fatal error: vector: No such file or directory

JingchiZhang, 2020-11-04:
Hi. I need some help with the software emulation. I think I follow the instructions clearly for each step and I export the all the path/ However when I run the software emulation "aarch64-linux-gnu-g++ -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I${SYSROOT}/usr/include/xrt -L${SYSROOT}/usr/lib -lOpenCL -lpthread -lrt -lstdc++ --sysroot=${SYSROOT}", it keeps tell me:

../../src/host.cpp:33:10: fatal error: vector: No such file or directory
   33 | #include <vector>
      |          ^~~~~~~~
compilation terminated.

I am not sure why this happen, vector should be included in g++ library. Am I missing something here?

tcxxxx, 2020-11-12:
I ran into the problem of missing the `bits/c++config` header previously. 
My solution was to first use the `locate` command to find the missing header file and then include the corresponding path in the make command. 

Also, it should be ensured that the variable SYSROOT is set correctly.

GkyHub, 2020-12-16:
I met the same problem.

Change SYSROOT to the extracted root file system, where there are folders like usr, root, lib, .... SYSROOT is where the compiler finds the necessary headers. I think the tutorial should give more information on how to set the variables ROOTFS and SYSROOT. 

This is a little bit strange because I thought that sysroot should be the one I downloaded from the Xilinx website. But actually here it should be the extracted rootfs. I am not sure if this is the right way to solve the problem but it works.

closed 2021-02-26 by rwarmstr