Xilinx/Vitis-Tutorials #17: Specific instructions on setting up SYSROOT and ROOTFS

tcxxxx, 2020-11-10:
Hi, 

Could you provide more specific instructions on setting up the variables SYSROOT and ROOTFS? 
The [documentation](https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/rvu1542160683426.html) is pretty vague on how to proceed. 

For example, to set up the ZCU102 platform, I have downloaded the ZYNQ common image on this [page](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html), and executed `sdk.sh` as instructed, but what is the correct way to set up SYSROOT and ROOTFS after this step? 

Thanks a lot!

imrickysu, 2020-11-11:
If you're using Vitis IDE, Vitis asks for SYSROOT, ROOTFS and Kernel when you create a system application.
https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/creatingvitisideproject.html#ariaid-title3 
Step 5

If you're using command line flow, you'll need to follow the makefile definition. For example in **_Vitis_Accel_Example -> Hellow World_**
https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/hello_world/Makefile defines it with the variable `EDGE_COMMON_SW=<rootfs and kernel image path>`

In Vitis_Library example, 
https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L2/tests/cvtcolor/bgr2hsv/Makefile
SYSROOT needs to be defined as an environment variable

Please note SYSROOT needs to point to aarch64-xilinx-linux, or cortexa9t2hf-neon-xilinx-linux-gnueabi directory

Please check more details with [Using Vitis Embedded Platforms ](https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/usingembeddedplatforms.html)

tcxxxx, 2020-11-12:
> If you're using Vitis IDE, Vitis asks for SYSROOT, ROOTFS and Kernel when you create a system application. [...]

Thank you very much!!

closed 2020-11-12 by tcxxxx