Xilinx/Vitis-Tutorials #18: Can't run in hw with examples of getting started with rtl kernels

La-tale, 2020-11-17:
I follow the tutorial of getting started with RTL kernels.
It works fine until building and running example of tutorials in Emulation-HW.
Also it works fine for building example in HW.
However when I run example in HW, it get stucked like below picture.
[image]

So I should do xbutil reset for running again this example.

I test various things, and I found that it works fine at ap_ctrl_hls interface, but it didn't work at ap_ctrl_chain interface which this example uses.

Did anyone who follow this example experience this problem?
If yes, how can solve this problem, and what is the cause?

Thank you!

randyh62, 2020-11-23:
ap_ctrl_chain is not supported in the RTL kernel. ap_ctrl_hs is the right protocol. We will correct this in the tutorial. Thank you.

La-tale, 2020-11-25:
Is it kernel problem??
Then could you let me know when ap_ctrl_chain is used?

randyh62, 2020-11-25:
ap_ctrl_chain is the default for C/C++ kernels in Vitis compiler and Vitis HLS. In this tutorial, the RTL code only contains the signals for ap_ctrl_hs as described under _Hardware Interface Requirements_ in the [Introduction](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow). If your RTL code has the signals to support ap_ctrl_chain, you can use that protocol on the kernel. Some of this is discussed in [XRT Execution Models](https://xilinx.github.io/XRT/2020.1/html/xrt_kernel_executions.html).

closed 2021-02-26 by rwarmstr