Xilinx/Vitis-Tutorials #19: Missing step in packaging IP documentation

TenzinCHW, 2020-11-18:
In the `Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow/package_ip.md` documentation, I have reached the "Check Integrity, Assign Properties, and Package IP" step where step 2 tells me to run the following commands in the Tcl console to "set these properties" with no reference to what "these properties" might mean.
```
set_property xpm_libraries {XPM_CDC XPM_MEMORY XPM_FIFO} $core
set_property sdx_kernel true $core
set_property sdx_kernel_type rtl $core
```

Also, the above lines result in the error `can't read "core": no such variable`.

Is there something I need to set before running these lines? And also which file should I run `check_integrity` on?

randyh62, 2020-11-18:
You can change $core to [ipx::current_core] in your command lines, or use this command line prior to the lines in your example:
set core [ipx::current_core]

I will correct this in the published version.

TenzinCHW, 2020-11-19:
Alright thank you.

Can I ask how is `check_integrity` used? I tried it on the component.xml in the IP directory but it fails the integrity check.

Thank you very much for the help. I must also add that the documentation in the READMEs are well done because they give users step-by-step examples to follow and get started quickly, while the user guides seems to have much more information that may be difficult to follow because there are a lot of side notes for extra options that are only needed in projects that use those functionalities. So thank you for the step-by-step guides!

TenzinCHW, 2020-11-19:
Ok nvm I'm just missing the `ipx` in front of everything. For those who are looking for the answer the command will be `ipx::check_integrity [ipx::current_core]`

closed 2021-02-20 by rwarmstr