Xilinx/Vitis-Tutorials #21: HW_EMU (hardware emulation) errors in the Cholesky example (likely caused by clSetKernelArg)

tcxxxx, 2020-11-24:
In the Cholesky example here ([`Vitis-Tutorials/Hardware_Accelerators/Introduction/03-Algorithm_Acceleration/docs/module1_baseline`](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Hardware_Accelerators/Introduction/03-Algorithm_Acceleration/docs/module1_baseline)), running `make run TARGET=hw_emu` will fail. Meanwhile, running `make run TARGET=sw_emu` will succeed. 

Specifically, the error information is: 

```
[XRT] WARNING: Argument '1' of kernel 'cholesky_kernel' is allocated in memory bank 'bank0'; compute unit 'cholesky_kernel_1' cannot be used with this argument and is ignored.
[XRT] ERROR: kernel 'cholesky_kernel' has no compute units to support required argument connectivity.
ERROR: clSetKernelArg() for kernel "cholesky_kernel", argument index 1. 
```

The error is caused by [this line](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Hardware_Accelerators/Introduction/03-Algorithm_Acceleration/docs/module1_baseline/src/test.cpp#L177), and it seems to be related to memory allocation/transfer and/or kernel argument setup, potentially function `clSetKernelArg`.

Could anyone help? Any suggestion/hint would be appreciated!

Environment: 
OS: Ubuntu 18.04; Device: xilinx_u200_xdma_201830_2; XRT build version: 2.8.0. 

A screenshot of the error information is shown below. 
[image]

tcxxxx, 2020-11-24:
As a workaround to the problem, I changed [this line](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Hardware_Accelerators/Introduction/03-Algorithm_Acceleration/docs/module1_baseline/src/test.cpp#L153) to `mext_io[0].flags = XCL_MEM_DDR_BANK1;`, and it worked: 

[image]

The problem was that the instance was by default connected to DDR bank1 even though in the code it was specified to connect to bank0. 
[image]

tcxxxx, 2020-11-24:
> As a workaround to the problem, I changed [this [...]

I wonder why in the HW_EMU mode it by default uses DDR bank1 instead of DDR bank0?

tcxxxx, 2020-11-24:
I wonder why, in the HW_EMU mode, it by default uses DDR bank1 instead of DDR bank0, even when the code sets the XCL_MEM_DDR_BANK0 flag?

randyh62, 2020-11-24:
I have run both sw_emu and hw_emu for the original code and was able to run both without problem. Can you retest the original files to see if there is another cause to your difficulties? 

[image]

tcxxxx, 2020-11-24:
@randyh62 Thank you! However, I am still running into the same issue when doing that in a completely new repo. 
My environment: ubuntu 18.04, g++ 7.5.0, XRT build version: 2.8.0. What is your environment? 
Maybe the problem only exists in certain environments.

[image]

The root cause is still the same, i.e., it tries to use BANK1 instead of BANK0.

closed 2020-12-16 by tcxxxx