Xilinx/Vitis-Tutorials #27: [HLS] Disable Automatic Loop Optimizations

tcxxxx, 2020-12-16:
As mentioned [in this tutorial here](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Hardware_Accelerators/Introduction/03-Algorithm_Acceleration/docs/module2_pipeline), 'simple loops and inner loops (for nested loops) are automatically pipelined by the tool'. 

I hope to know is there is any reference to the documentation about the pipelining optimizations automatically applied?
Is it possible to disable such optimizations, e.g., maybe by changing some compiler configurations? 

Thanks!

f-rivo, 2021-02-18:
The PIPELINE option can be turned off locally, at the loop level using #pragma HLS pipeline off.
See more information here: https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/hlspragmas.html#fde1504034360078.
Besides the web based documentation, the option is also described in the user guide: https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug1399-vitis-hls.pdf

tcxxxx, 2021-02-18:
> The PIPELINE option can be turned off locally at the loop level using #pragma HLS pipeline off.
> See more information here: [...]

Thank you very much the notes! 
As my concern with this is fully resolved now, I also noticed [this page](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/vitis_hls_process.html#djn1584047476918) contains several default behaviors (including default loop optimizations) of Vitis HLS, which is very helpful.

closed 2021-02-18 by tcxxxx