Xilinx/Vitis-Tutorials #32: ERROR when build xo /03-rtl_stream_kernel_integration

ykgot, 2021-01-07:
When I tried to build xo with the following comment for 03-rtl_stream_kernel_integration project, it had some ERRORs.
How to solve this?
Thanks!

make all_xo TARGET=hw PLATFORM=xilinx_u200_xdma_201830_2

ERROR: [v++ 214-160] in function 'alpha_mix': Local stream 'mixed_img.4' has conflict depth setting (xx/03-rtl_stream_kernel_integration/hw/alpha_mix.cpp:46:61)
ERROR: [v++ 214-160] in function 'alpha_mix': Local stream 'resized_time_img.4' has conflict depth setting (xx/03-rtl_stream_kernel_integration/hw/alpha_mix.cpp:44:61)

fmanuelsg, 2021-01-20:
same issue on a working project build with 2020.1

rwarmstr, 2021-02-20:
These tutorials have this issue if you run 2020.1 tutorials on 2020.2 with Vitis HLS, the repos have since been updated. Please let us know if you still see this.

manuel-sanchez-nttdisruption, 2021-02-20:
2020.2

rwarmstr, 2021-02-26:
You'll see the issue if you build the 2020.1 tutorial branch on 2020.2. You should not see the issue if you build the 2020.2 branch on 2020.2. We also committed a patch yesterday against the 2020.1 branch that should resolve it, but please let us know if you continue to see difficulties.

closed 2021-08-18 by Premduth