Xilinx/Vitis-Tutorials #35: Issue: cannot find "Enable platform interfaces"

Tarekro, 2021-01-15:
hi, 

On point 5 in "Create Base Vivado Project from Preset" under the tutorial "Vitis-Tutorials/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step1" I cannot find the "Platform interfaces" in the Window panel on my VIVADO 2020.2. Is someone has the same problem and he already fixed it? Any idea to find the platform window? I attached a screenshot of my environment. 
[image]

imrickysu, 2021-01-21:
In Vivado 2020.2, "Platform Interfaces" is changed to "Platform Setup". But it's only available if your project is "extensible Vitis platform project" type. It can be enabled when you are creating a project, or after the project is created, select Settings in Flow Navigator -> General -> enable "Project is an extensible Vitis Platform".

We will publish 2020.2 version of this tutorial soon. Stay tuned.

imrickysu, 2021-01-25:
Hi @Tarekro , 
2020.2 version is updated. Please check: https://github.com/Xilinx/Vitis-Tutorials/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step1.md

Tarekro, 2021-01-25:
Hi @imrickysu 

Thanks for your help, I will check this new version;

closed 2021-02-18 by rwarmstr