Xilinx/Vitis-Tutorials #37: 07-yolov4-tutorial/scripts > ./compile_yolov4.sh error

hopesala, 2021-01-18:
hello, I am using  vitis-ai 1.3 to try the tutorial 07-yolov4-tutorial, but when I execute compile_yolov4.sh the error bellow occurred, what could be the reason and how to solve it? thanks very much~

before the step execute compile_yolov4.sh, I have generated tf_model.pb and yolov4_quantized folder(containing deploy_model.pb quantize_eval_model.pb ) successfally.
besides, I have changed the compile_yolov4.sh because the vitis-ai1.3 has changed json file path accordingly.

TARGET=ZCU102
NET_NAME=yolov4

**ARCH=${CONDA_PREFIX}/arch/DPUCZDX8G/${TARGET}/arch.json**

(vitis-ai-tensorflow) Vitis-AI /workspace/888888/07-yolov4-tutorial/scripts > ./compile_yolov4.sh 
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['../yolov4_quantized/deploy_model.pb'], model_type='tensorflow', out_filename='../yolov4_compiled//yolov4_org.xmodel', proto=None)
[INFO] tensorflow model: ../yolov4_quantized/deploy_model.pb
[INFO] parse raw model     :  0%| | 1/407 [00:00<00:00, 1246.45it/s]            
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 243, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 1847, in __generate_xmodel
    ), f"[ERROR] TF Conv2d requires two inputs: actual: {bottom}."
AssertionError: [ERROR] TF Conv2d requires two inputs: actual: ['image_input'].

hopesala, 2021-01-18:
@rwarmstr can you help me find the possible reson for this error, big thanks~

shua1zhang, 2021-01-20:
Hi @hopesala , we will take a look on this. Thank you for the understanding.

shua1zhang, 2021-03-03:
Hi @hopesala, sorry for the delayed reponse. Engineering team is still looking into the root cause and will let you know once finished. 
Thank you for the understanding.

shua1zhang, 2021-03-03:
Hi @hopesala , 
We notice there is some issue with the version of the tutorial as it's still using the vai 1.2. We will let author to update accordingly. 
if you want to compile the model correctly in vai 1.3, the compiler will need to use the "quantized eval pb" file genereated by quantizer. 
Also please check the new arch file path in VAI 1.3 docker according to UG1414. 
Thank you.

Premduth, 2021-08-18:
Please try this in

https://github.com/Xilinx/Vitis-AI-Tutorials

All ML tutorials moved there

closed 2021-08-18 by Premduth