Xilinx/Vitis-Tutorials #40: Segmentation faults on cpu_src in Hardware_accelerators/Introduction/03

andreaf2405, 2021-02-09:
I get a segmentation fault at runtime on the baseline CPU program. Any clues?

rwarmstr, 2021-02-20:
This is fairly common on machines with limited stack sizing, we just applied a patch to the repo to address it. Can you please try again?

rwarmstr, 2021-02-26:
Hi @andreaf2405, since we haven't heard back from you and we believe the issue is resolved by commit b98e24bbf847778672cd5cc8f41cc4a60cede566 I'm going to go ahead and close this issue, but if you still have trouble please feel free to reopen it!

closed 2021-02-26 by rwarmstr

andreaf2405, 2021-02-26:
> Hi @andreaf2405, since we haven't heard back from you and we believe the issue is resolved by commit [b98e24b](https://github.com/Xilinx/Vitis-Tutorials/commit/b98e24bbf847778672cd5cc8f41cc4a60cede566) I'm going to go ahead and close this issue, but if you still have trouble please feel free to reopen it!

Thanks a lot for addressing it. I haven't had the chance to test it as of yet - I will get back to you in case it is still failing!