Xilinx/Vitis-Tutorials #41: Help/intuition debugging vadd example (Getting_Started/Vitis), zocl-drm amba:zyxclmm_drm: swiotlb buffer is full

vmayoral, 2021-02-17:
Greetings,

First let me credit authors and maintainers for this repository. From the top of the list cheers for @xlinx-dachang, @rwarmstr and @f-rivo among others. Well done. I found the material very useful and the quality rather high. I'm excited about these tutorials and I believe many share the same vision. Maybe I'll get a chance to contribute in the future myself with some additional tutorials :).

I'm launching here a request for help/pointers debugging an issue that probably some other may have encountered (or will). The accelerated kernels I'm developing are slowly building on top of the examples provided. I found [`vadd` one](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md) incredibly illustrative so I'll use this one to motivate the problem:

The PS-side of the code uses a `DATA_SIZE` pre-processor definition to determine the `size` of the vectors to be used. The example uses a default size of `4096`. If (like me) you happen to be dealing with data-structures a bit more dense than this, you'd typically try other values:

```C
//#define DATA_SIZE 4096  // 2**12
#define DATA_SIZE 65536  // 2**16
//#define DATA_SIZE 262144  // 2**18
```

While testing with the `hw_emu` build target (see [build targets](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part1.md)), `4096` and `65536` produce the expected results but when testing with `262144`, the following occurs:

```bash
devmachine# ./app.exe.262144
INFO: Add accelerated vector example
INFO: Found Xilinx Platform
[  236.229053] [drm] Pid 828 opened device
INFO: Loading 'vadd.xclbin'
[  236.692526] [drm] zocl_xclbin_read_axlf The XCLBIN already loaded
[  236.692575] [drm] zocl_xclbin_read_axlf 66bc6e53-9c1d-4127-b026-d1f6829206f7 ret: 0
[  236.705650] [drm] bitstream 66bc6e53-9c1d-4127-b026-d1f6829206f7 locked, ref=1
[  236.706336] [drm] Reconfiguration not supported
[  236.708989] [drm] bitstream 66bc6e53-9c1d-4127-b026-d1f6829206f7 unlocked, ref=0
[  236.725520] [drm] bitstream 66bc6e53-9c1d-4127-b026-d1f6829206f7 locked, ref=1
[  236.753934] zocl-drm amba:zyxclmm_drm: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 408 (slots)
[  236.756095] zocl-drm amba:zyxclmm_drm: overflow 0x0000000873069000+1048576 of DMA mask ffffffff bus mask 0
[  236.756350] ------------[ cut here ]------------
[  236.756687] WARNING: CPU: 3 PID: 828 at kernel/dma/direct.c:35 report_addr+0x38/0xa0
[  236.757377] Modules linked in: zocl(O)
[  236.757842] CPU: 3 PID: 828 Comm: app.exe.262144 Tainted: G        W  O      5.4.0-xilinx-v2020.2 #1
[  236.758102] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
[  236.758233] pstate: 80000005 (Nzcv daif -PAN -UAO)
[  236.758359] pc : report_addr+0x38/0xa0
[  236.758439] lr : report_addr+0x98/0xa0
[  236.758535] sp : ffff800011b0bb10
[  236.758619] x29: ffff800011b0bb10 x28: ffff800011b0bcf8
[  236.758914] x27: 0000000000000018 x26: ffff00087ab0fc00
[  236.759088] x25: ffff0008793d0780 x24: 0000000000000000
[  236.759227] x23: 0000000000000000 x22: ffff00087ab4dc10
[  236.759363] x21: 0000000000100000 x20: ffff00087a31b140
[  236.759443] x19: ffff00087ab4dc10 x18: 0000000000000010
[  236.759524] x17: 0000000000000041 x16: 0000000000000000
[  236.759662] x15: ffff00087a31b568 x14: 616d20414d442066
[  236.759818] x13: 6f20363735383430 x12: 312b303030393630
[  236.760174] x11: 3337383030303030 x10: 3030783020776f6c
[  236.760318] x9 : ffff800011193000 x8 : 0000000000000220
[  236.760464] x7 : 0000000000000006 x6 : 0000000000000001
[  236.760610] x5 : 0000000000000000 x4 : 0000000000000001
[  236.760778] x3 : 0000000000000006 x2 : 9ca27b1d401e1900
[  236.760946] x1 : 9ca27b1d401e1900 x0 : 0000000000000000
[  236.761277] Call trace:
[  236.761342]  report_addr+0x38/0xa0
[  236.762441]  dma_direct_map_page+0x110/0x130
[  236.762571]  dma_direct_map_sg+0x78/0xe0
[  236.763016]  zocl_userptr_bo_ioctl+0x16c/0x2c8 [zocl]
[  236.763185]  drm_ioctl_kernel+0xb8/0x108
[  236.763408]  drm_ioctl+0x200/0x410
[  236.763495]  do_vfs_ioctl+0x878/0xa20
[  236.763625]  ksys_ioctl+0x44/0x90
[  236.763738]  __arm64_sys_ioctl+0x1c/0x28
[  236.763849]  el0_svc_common.constprop.0+0x68/0x160
[  236.763976]  el0_svc_handler+0x6c/0x88
[  236.764049]  el0_svc+0x8/0xc
[  236.764111] ---[ end trace e731bc0ce1c1afd2 ]---
[  236.764900] [drm:zocl_userptr_bo_ioctl [zocl]] *ERROR* Map SG list failed
[  236.796800] zocl-drm amba:zyxclmm_drm: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 408 (slots)
[... 100 lines cut ...]
[  236.763408]  drm_ioctl+0x200/0x410
[  236.763495]  do_vfs_ioctl+0x878/0xa20
[  236.763625]  ksys_ioctl+0x44/0x90
[  236.763738]  __arm64_sys_ioctl+0x1c/0x28
[  236.763849]  el0_svc_common.constprop.0+0x68/0x160
[  236.763976]  el0_svc_handler+0x6c/0x88
[  236.764049]  el0_svc+0x8/0xc
[  236.764111] ---[ end trace e731bc0ce1c1afd2 ]---
[  236.764900] [drm:zocl_userptr_bo_ioctl [zocl]] *ERROR* Map SG list failed
[  236.796800] zocl-drm amba:zyxclmm_drm: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 408 (slots)
[  236.797746] [drm:zocl_userptr_bo_ioctl [zocl]] *ERROR* Map SG list failed
[  236.819922] zocl-drm amba:zyxclmm_drm: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 408 (slots)
[  236.821733] [drm:zocl_userptr_bo_ioctl [zocl]] *ERROR* Map SG list failed
Resetting rate control (66261 samples)
Resetting rate control (65563 samples)
TEST PASSED
[  493.960032] [drm] bitstream 66bc6e53-9c1d-4127-b026-d1f6829206f7 unlocked, ref=0
[  493.989450] [drm] Pid 828 closed device
```

</details>

This is certainly not the expected behavior right? The results lead to the following questions which I hope some of the authors/maintainers can help clarify:
- Is this the expected behavior? The accelerated kernel is using `const unsigned int`. AFAIK and out of `#include <limits>`, `UINT_MAX` is `4294967295` which is higher than `262144`. Am I missing something?
- How can I engage into debugging this? Intuition tells me that I should start looking at `zocl` but frankly, I don't have enough experiences yet to have a strong opinion. Can someone more experienced split this to me into *reasonable steps he/she would take to debug this*?

vmayoral, 2021-02-17:
In case it isn't obvious, reproducing the errors above should be pretty straightforward, it's just a matter of modifying https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/example/src/host.cpp#L31 with the values discussed above.

rwarmstr, 2021-02-18:
Hi @vmayoral , thanks for the feedback! I'll look into it but don't suspect initially that this is a problem with the driver, it may lie somewhere else. Just to double check, and in case there's any interaction, it looks like you're running this on an embedded platform. May I ask which one?

Thanks,
Rob

vmayoral, 2021-02-18:
Thanks for the reaction @rwarmstr! I'm using the ZCU102 but all the above was tested on emulation with the `hw_emu` build target.

rwarmstr, 2021-02-24:
Hi @vmayoral , wanted to quickly check in with an update. It seems you've hit a genuine bug and it's going to take us a bit longer to debug than I thought! I'll keep you updated here; it may be a quick fix but we haven't gotten down to the root cause yet. For hardware emulation of embedded systems we have many components interacting (QEMU, the hardware simulator, XRT, etc) and it can sometimes take a little while to track down.

rwarmstr, 2021-02-25:
So, it turns out that the root cause here is kind of interesting (at least to me!). We're using the same memory allocation code in the host example for both data center and embedded devices, with the flag CL_MEM_ALLOCATE_HOST_PTR passed into the constructor for the cl::Buffers. On embedded systems, what I thought would happen is that it would allocate from the CMA pool whether you use that flag or not, but it turns out that using that flag allocates a standard userspace pointer for the buffer.

The kernels themselves require contiguous memory, so when you enqueue the buffer migration behind the scenes what will happen is that the DMA engine in the Zynq hardware will build a scatter-gather list to copy your paged data into a new memory region allocated through the CMA. It's during this stage of the operation that you were seeing the kernel taint.

So, there are a few things:

1) You don't want to allocate memory with that flag, because it's adding an inefficient extra DMA operation. Fortunately, that's easy to fix, we can just remove it.

2) The kernel taint is actually benign, but we need to improve the error handling in that code path. We're making that change to the XRT code and will release the fix in our 2021.1 software update.

3) When you increase the length of the vectors, the emulation time increases. That may seem obvious but even with a relatively small 1MB buffer it takes a couple minutes to run. Maybe you took that for a hang? The test does eventually pass, but it's performing a gate-level logic simulation of the hardware behind the scenes which isn't super fast. :) 

I have pull request #45 ready to merge to address points one and three, and for point two we'll get it fixed in the next release. Hope that helps!

vmayoral, 2021-02-26:
Thanks a lot @rwarmstr for the reasoned answer. I made the suggested modifications and reproduced again the issues I was experiencing Tests pass. Provided I'm patient enough (262144 and higher gets you waiting a bit 😭 ) I can analyze the results which unblocks me.

> Maybe you took that for a hang? The test does eventually pass, but it's performing a gate-level logic simulation of the hardware behind the scenes which isn't super fast. :)

Yeah, my bad. That was indeed my assumption.

> I have pull request #45 ready to merge to address points one and three, and for point two we'll get it fixed in the next release. Hope that helps!

Fantastic. Looking forward for next release then! All issues dealt from my end so feel free to close this ticket.

rwarmstr, 2021-02-26:
Thanks! I'll close it.

closed 2021-02-26 by rwarmstr