Xilinx/Vitis-Tutorials #42: Quantize with vitis-ai-pytorch and met 'parsing failed(unsupported)' in my torch.aotograd.Function derived class

beenleyi, 2021-02-18:
I am doing a new deep learning algorithm research with my own torch.aotograd.Function derived class, but I met 'parsing failed(unsupported)' when quantizing with vitis-ai-pytorch. Does vitis-ai-pytorch support users' own autograd class, or does vitis-ai-pytorch support another class except the model class? 

I got the following info:

```
(vitis-ai-pytorch) Vitis-AI /workspace/workplace/BP > python -u quantize.py --quant_mode calib 2>&1 | tee quant_calib.log
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[NNDCT_NOTE]: Loading NNDCT kernels...

-----------------------------------------
PyTorch version : 1.4.0
3.6.12 |Anaconda, Inc.| (default, Sep 8 2020, 23:10:56)
[GCC 7.3.0]
-----------------------------------------
Command line options:
--dset_dir : dataset
--float_model : float_model
--quant_mode : calib
--batchsize : 100
--quant_model : quant_model
-----------------------------------------
No CUDA devices available..selecting CPU

[NNDCT_WARN]: CUDA is not available, change device to CPU

[NNDCT_NOTE]: Quantization calibration process start up...

[NNDCT_NOTE]: =>Quant Module is in 'cpu'.

[NNDCT_NOTE]: =>Parsing SCNN...
aten_op 'ActFun' parse failed(unsupported)
```

And part of my code:

```python
class ActFun(torch.autograd.Function):

    @staticmethod

    def forward(ctx, input):

        ctx.save_for_backward(input)

        return input.gt(thresh).float()

    @staticmethod

    def backward(ctx, grad_output):

        input, = ctx.saved_tensors

        grad_input = grad_output.clone()

        temp = abs(input - thresh) < lens

        return grad_input * temp.float()

act_fun = ActFun.apply

# membrane potential update

def mem_update(ops, x, mem, spike):

    mem = mem * decay * (1. - spike) + ops(x)

    spike = act_fun(mem) # act_fun : approximation firing function

    return mem, spike
```

I call mem_update in my forward function.

My quantize.py is almost the same as quantize.py (github: Vitis-Tutorials/Machine_Learning/Design_Tutorials/09-mnist_pyt_master/files/quantize.py)

 

Thank you so much.

rwarmstr, 2021-02-18:
Hi @beenleyi , the best person I have on the team to answer this is currently out of the office for the lunar new year, if you can wait a bit I'm happy to have them get back to you.

beenleyi, 2021-02-20:
Thank you so much, @rwarmstr .

I found that torch.gt is not supported too. I got the following info:
```python
aten_op 'gt' parse failed(unsupported)
```
Any method to solve it?

Thank you all.

niuxjxlnx, 2021-02-22:
So far vai_q_pytorch does not support customized OP and function. But it was in plan of coming release.

beenleyi, 2021-02-23:
Thank you! @niuxjxlnx 
How about the autograd derived class problem? Any information is of great help to me.

niuxjxlnx, 2021-02-24:
It is not in short term plan. Simple customized OP is base of it. And futher enhancement will be planned after the base become solid enough.

beenleyi, 2021-02-26:
Thank you so much.@niuxjxlnx

closed 2021-02-26 by beenleyi