Xilinx/Vitis-Tutorials #47: Pre-synthesis Control Flow option not available when running C-simulation

ThomasvdWerff, 2021-03-03:
While working on the synth and analysis part of the Vitis HLS getting started tutorial I noticed that one of the options has been removed from the C simulation dialog. See the screenshot attached for comparison. 

Tutorial:
[image]

Current:
[image]

Is this option (re)moved?

randyh62, 2021-03-03:
The images were taken from the Vitis HLS 2020.2 tool. Sometimes the dialog box is too small to display all the options. Can you see if you can expand the dialog box by making it longer?

ThomasvdWerff, 2021-03-03:
That's correct. I've tried expanding it as well but without any positive results. I've also tried to select the "ghost"-button a few times but that did not work either. It seems like it really disappeared from the dialog.

randyh62, 2021-03-03:
[image]

Here you can see I am running 2020.2 and it shows up. I don't think there is a setting to disable the option, but I will check into it a little further. Can you say what OS you are working with?

ThomasvdWerff, 2021-03-03:
Windows 10 Education, build 19042. Also thanks for your time.

randyh62, 2021-03-04:
Still verifying, but it looks like it might be a limitation of the Windows build.

ThomasvdWerff, 2021-03-04:
I've received confirmation of the same problem on Windows 10 Pro, build 18363.  Both installations are no longer than a week or two old. On the education OS, I've also activated all the possible licenses. On the Pro version, only the following are activated: "Analyzer, HLS, V_WebPACK"

randyh62, 2021-03-04:
I have confirmed this is a limitation of the Windows version. I will add this to the tutorial instructions. However, it looks like you will have to skip this part of the tutorial.  I will see if there are any other limitations we should be aware of, but please let us know of any other issues you may encounter.

Premduth, 2021-08-18:
Closing as per understanding it is a known issue

closed 2021-08-18 by Premduth