Xilinx/Vitis-Tutorials #50: File not found common/xf_common.hpp when building Runtime_and_System_Optimization/Introduction/design_source/hw_src

HFTrader, 2021-03-14:
```
===>The following messages were generated while  performing high-level synthesis for kernel: resize_accel_rgb Log file: /ctc/Xilinx/Vitis-Tutorials/Runtime_and_System_Optimization/Introduction/design_source/hw_src/_x/resize_rgb/resize_accel_rgb/vitis_hls.log :
ERROR: [v++ 207-812] 'common/xf_common.hpp' file not found: /ctc/Xilinx/Vitis-Tutorials/Runtime_and_System_Optimization/Introduction/design_source/hw_src/resize_rgb.cpp:32:10
ERROR: [v++ 60-300] Failed to build kernel(ip) resize_accel_rgb, see log for details: /ctc/Xilinx/Vitis-Tutorials/Runtime_and_System_Optimization/Introduction/design_source/hw_src/_x/resize_rgb/resize_accel_rgb/vitis_hls.log
ERROR: [v++ 60-773] In '/ctc/Xilinx/Vitis-Tutorials/Runtime_and_System_Optimization/Introduction/design_source/hw_src/_x/resize_rgb/resize_accel_rgb/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-812] 'common/xf_common.hpp' file not found: /ctc/Xilinx/Vitis-Tutorials/Runtime_and_System_Optimization/Introduction/design_source/hw_src/resize_rgb.cpp:32:10
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [resize_rgb.xo] Error 1
```

HFTrader, 2021-03-14:
I just tried switching to the 2020.1 branch but the result is the same.

rwarmstr, 2021-03-14:
Do you see any files under the hw_src/Vitis_Libraries directory? If not, I suspect you may not have cloned with the --recurse-submodules flag and the Vitis Libraries aren't downloaded. You should be able to resolve it by running:

`git submodule init`

followed by

`git submodule update`

Please let me know if that works.

HFTrader, 2021-03-14:
You were correct. I did not provide the --recurse-submodules argument to the git clone command line.

closed 2021-03-14 by HFTrader