Xilinx/Vitis-Tutorials #53: Accracy drops to 0.1 on ZCU104 : Machine_Learning/Introduction/03-Basic/Module_4/DenseNetX

satoru-i, 2021-03-17:
Hi, I'm trying the tutorial of Machine_Learning/Introduction/03-Basic/Module_4.

After execution the command vai_c_tensorflow at the step6, copied the densenetx.model to ZCU104 board.
I executed 'python3 app_mt.py' at the step 7, the accuracy drops to 0.1 on the ZCU104 board.
-----------------------------------------------------------------
root@xilinx-zcu104-2020_2:~/target# python3 app_mt.py
Command line options:
 --image_dir :  images
 --threads   :  1
 --model     :  model_dir/densenetx.xmodel
Pre-processing 10000 images...
Starting 1 threads...
FPS=509.41, total frames = 10000 , time=19.6307 seconds
output buffer length: 10000
Correct: 1000 Wrong: 9000 Accuracy: 0.1
root@xilinx-zcu104-2020_2:~/target#
-----------------------------------------------------------------

I changed the board to ZCU102, but the result was the same.

The accuracy results of step3(frozen_graph.pb) and step5(quantize_eval_model.pb) seems good, below.
frozen_graph: 0.9306
quantize_eval_model: 0.9265

What could be the reason of the accuracy drop on the board?

fanz-xlnx, 2021-03-22:
Hi Satoru,
Thanks for the question. The 0.1 is not the actual accuracy but due to a library mismatch in app_mt.py. The file will be updated and push to Github later today.

satoru-i, 2021-03-25:
Hi fanz,
Thank you for updating the file.
I replaced the app_mt.py and executed on ZCU104 , but accuracy didn't change, 0.1.
Should I change some libraries or start over from the previous step?

fanz-xlnx, 2021-03-29:
Hi Satoru,

There is no need to change the library. I don't see this issue on my side. Would you please send me the quantize_eval_model.pb generated in Step 5 so I can reproduce the issue. Thanks. The file could be send to fanzhang@xilinx.com

qiyangjie, 2021-05-25:
Hi fanz,

I think I has the same issue as Satoru. Do you have any update solution for that? @fanz-xlnx 
Thanks

fanz-xlnx, 2021-05-26:
Hi Yangjie,

The issue could result from the xmodel differentiation. Can I have your email address?

qiyangjie, 2021-05-26:
Sure. My email address is qiy001@udayton.edu.

> Hi Yangjie,
> 
> The issue could result from the xmodel differentiation. Can I have your email address?

Rivon8, 2021-07-26:
Hi everyone,

I have the same problem. Is there anywhere the final solution for that?
Thanks

Premduth, 2021-08-18:
Please try this in

https://github.com/Xilinx/Vitis-AI-Tutorials

All ML tutorials moved there

closed 2021-08-23 by Premduth