Xilinx/Vitis-Tutorials #54: Compile yolov4 with custom data set output no DPU graph

quyetvk, 2021-03-19:
Hello Xilinx's member

I tried to follow tutorial "Vitis-Tutorials/Machine_Learning/Design_Tutorials/07-yolov4-tutorial/" to execute Yolov4 to Ultra96v2.
I used custom dataset. I found that there is no graph in DPU.

```
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210319-091311-818"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: deploy, with op num: 693
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 2, DPU subgraph number 0
[UNILOG][INFO] Compile done
```

Could you please tell me the reason?
Thank you very much!

shua1zhang, 2021-03-21:
Hi @quyetvk,
This issue is new and we haven't never met the similar case. Is it possile to proivde quantized model for us to check?

shua1zhang, 2021-03-21:
Hi @quyetvk,

I tried to figure out what could be the possible root cause.

I guess you are using Vitis AI 1.3 version and with Caffe conversion flow, is that correct? 

As the tutorial is still in Vitis AI 1.2 version and update is undergoing, the quantization command here misses "-keep_fixed_neuron" flag which is newly required by 1.3 xmodel compiler. Please refer to [UG1414 caffe quantization](https://www.xilinx.com/html_docs/vitis_ai/1_3/caffe.html#ariaid-title3)

Please have test at your side.

quyetvk, 2021-03-22:
@shua1zhang Thank you for your feedback.

> Which conversion flow did you use? Is it possile to proivde quantized model for us to check?
I followed "3.2) Darknet Model Conversion" in https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial
I upload the quantized model and shere here
https://1drv.ms/u/s!An1u2J7R7eeOgREf-OF4LLMXoRrt?e=4imR5e

Just note here, the error of invalid prototxt occurred related to duplicated bottom field:
```
layer {
  name: "layer110-concat"
  type: "Concat"
  bottom: "layer105-conv"
  bottom: "layer107-conv"
  bottom: "layer105-conv"
  bottom: "layer104-conv"
  top: "layer110-concat"
  phase: TRAIN
}
```

So, we tried to remove one duplicated row to solve the error. Then the compiling is successful. Maybe there was problem in quantization step.

> I guess you are using Vitis AI 1.3 version and with Caffe conversion flow, is that correct?
Yes, it is correct.

> As the tutorial is still in Vitis AI 1.2 version and update is undergoing, the quantization command here misses "-keep_fixed_neuron" flag which is newly required by 1.3 xmodel compiler. Please refer to UG1414 caffe quantization
I tried to add keep_fixed_neuron option to quantization command (by modifying 
But other error occurred at compiling time:

```
[UNILOG][INFO] Begin to compile...

module_infer = 0, counter_m[module_infer] = 11, counter_p[module_infer] = 11
module_idx = 0, counter_m[module_idx] = 11, counter_p[module_idx] = 11
module_idx = 1, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 2, counter_m[module_idx] = 9, counter_p[module_idx] = 9
module_idx = 3, counter_m[module_idx] = 0, counter_p[module_idx] = 0
[UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.]
```

Do you have any idea?
Thanks for your support!

shua1zhang, 2021-03-22:
Hi @quyetvk,

By  "So, we tried to remove one duplicated row to solve the error. Then the compiling is successful. " you mean the compilation and execution on board are all successful with out adding "keep_fixed_neuron"? 

For the last issue, is it corrsponding to the quantized model you uploaded? 

Thank you.

quyetvk, 2021-03-22:
Hi @shua1zhang 

> By "So, we tried to remove one duplicated row to solve the error. Then the compiling is successful. " you mean the compilation and execution on board are all successful with out adding "keep_fixed_neuron"?

The compilation is successful without adding "keep_fixed_neuron" but there is no graph on DPU as I described.

> For the last issue, is it corrsponding to the quantized model you uploaded?

Yes, my upload for quantized model is for issue of no graph on DPU. 

When we added "keep_fixed_neuron", quantizing is successful but compiling is failed as previous report. I did not upload quantized model for this case. If you need it, I will upload.

Thank you!

shua1zhang, 2021-03-23:
Hi @quyetvk ,
Yes please also upload the model quantized with "keep_fixed_neuron" but failed to compile. 
Thank you.

quyetvk, 2021-03-24:
Hi @shua1zhang 

I upload the model quantized with "keep_fixed_neuron" here.
https://1drv.ms/u/s!An1u2J7R7eeOgRL_1JcvCsGVTbu4?e=BChe6g

Best regards

shua1zhang, 2021-04-13:
Hi @quyetvk ,
The tutorial is fully updated to VAI 1.3 version. The issue you met should all be solved.

quyetvk, 2021-04-14:
Hi @shua1zhang 
Thank you for your information.
But unfortunately, I have not yet compiled successfully.  I updated docker image to latest and use latest tutorial but I got the error when compiling:

```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['yolov4_quantized/deploy.caffemodel'], model_type='caffe', out_filename='yolov4_compiled/dpu_yolov4_voc_org.xmodel', proto='yolov4_quantized/deploy.prototxt')
[INFO] caffe model: yolov4_quantized/deploy.caffemodel
[INFO] caffe model: yolov4_quantized/deploy.prototxt
[INFO] parse raw model     :100%|███████████████████████████████████████████████████████████████████| 383/383 [00:10<00:00, 38.19it/s]
[INFO] infer shape (NCHW)  :100%|███████████████████████████████████████████████████████████████████| 383/383 [00:00<00:00, 7994.12it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████████████████████| 383/383 [00:00<00:00, 6552.56it/s]
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████████████████████| 383/383 [00:00<00:00, 651.84it/s]
[INFO] generate xmodel: /workspace/07-yolov4-tutorial/yolov4_compiled/dpu_yolov4_voc_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210414-082248-293"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: deploy, with op num: 823
[UNILOG][INFO] Begin to compile...
module_infer = 0, counter_m[module_infer] = 17, counter_p[module_infer] = 17
module_idx = 0, counter_m[module_idx] = 17, counter_p[module_idx] = 17
module_idx = 1, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 2, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 3, counter_m[module_idx] = 0, counter_p[module_idx] = 0
[UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.]
```

Some one said in "https://github.com/Xilinx/Vitis-AI/issues/346" the issue may be related to calib data in quantization. So, I am asking him. If you have idea, please let me know.

Best regards

Premduth, 2021-08-23:
Please reference here for ML tutorials:
https://github.com/Xilinx/Vitis-AI-Tutorials

If this issue still there, open with the latest version test.

closed 2021-08-23 by Premduth