Xilinx/Vitis-Tutorials #55: Path to platforms

m-kru, 2021-03-19:
In the introduction it is written:

> To specify the location of the target development platform for Alveo U50 accelerator card, set the following environment variable:
> 
> `export PLATFORM_REPO_PATHS=<path to platforms>`

What is the default path to platforms after default installation?
I wanted to create new Application Project, but there are no platforms to choose from:
[image]

imrickysu, 2021-03-21:
Default install path for Alveo platforms is /opt/xilinx/platforms
For embedded platforms, it can be extracted to any paths, including the path above. This path will be scanned by Vitis. If your platform is in this path, you won't need the PLATFORM_REPO_Paths environment variable.

m-kru, 2021-05-10:
@imrickysu I have installed the platform files for U50 to `/opt/xilinx/platforms/`. Now, I can choose this platform in `Xilinx Vitis` program, however I am not able to choose this board in the `Vitis HLS` program.

[image]

imrickysu, 2021-08-21:
Hi @m-kru , Vitis HLS board setup shares the Vivado board setup. Vitis HLS only needs device part info from the board setup. The contents in Alveo platform is not needed by Vitis HLS.
Please launch Vivado and go to Tools -> Vivado Store to download the boards

closed 2021-08-21 by imrickysu