Xilinx/Vitis-Tutorials #60: Core dump on AWS F1 running 02-MNIST_classification_tf example

danlg, 2021-04-01:
After having built everything successfully from step 0 to 7 from the tutorial ML/02-MNIST_classification_tf 

But I want to run the xmodel on  AWS F1 f1.2xlarge, I get the errow below. Can you please confirm U50 is ok to run on  f1.2xlarge ottherise which target ?

(vitis-ai-tensorflow) Vitis-AI /workspace/build/target_u50 > /usr/bin/python3 app_mt.py -m model_dir/customcnn.xmodel
Command line options:
 --image_dir :  images
 --threads   :  1
 --model     :  model_dir/customcnn.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0401 08:21:48.498520   440 dpu_controller.cpp:44] Check failed: !the_factory_methods.empty()
*** Check failure stack trace: ***
Aborted (core dumped)

danlg, 2021-04-01:
@imrickysu @stsoe @markharvey

Premduth, 2021-08-18:
@ThomasXilinx  can you have someone look at this?