Xilinx/Vitis-Tutorials #61: 线程问题

LinErDeng, 2021-04-02:
您好：
请问一下。为什么在使用./test_video_yolov3 dpu_yolov4_voc 0 -t4时，t无论是1还是4，检测的FPS都是一样的？

fanz-xlnx, 2021-04-06:
您好，
这个dpu_yolov4_voc是您自己的模型吗？
我用发布ZCU04镜像中的例程测试：./test_video_yolov3 yolov3_voc 0 -t 4 或者 ./test_video_yolov3 yolov4_leaky_spp_m 0 -t 4均未能复现问题。修改线程数1或者4对帧率有显著影响。请问您的测试平台是什么？能否再具体描述一下测试的流程？

LinErDeng, 2021-04-06:
1.我时用只含单个类别的数据集训练网络，得到cfg和weights文件；
2.通过您这个包进行了bash scripts/darknet_convert.sh以及后续的量化和编译过程，得到elf文件；
3.将dpu_yolov4_voc.prototxt中的类别数改为1之后和得到的elf文件拷贝到了板子ZCU104上；
4.将yolov3_voc_accuracy.cpp中的类别也改成我自己的那一个类别，然后bash -x build.sh
5.最后运行./test_video_yolov3 dpu_yolov4_voc 0 -t4这个命令，在执行这个命令时，线程位1和4没有区别。

fanz-xlnx, 2021-04-06:
Vitis AI Library提供了三个级别的API供不同场景的需求使用。
test_video_yolov3中使用的是API1，也是封装程度最高的API，主要用于快速demo和展示。我们不建议用户用这个级别的API进行定制模型部署。
API2可能更适合您的使用场景（相同的模型框架，不同的训练集和类别），您可以参考Vitis-AI/demo/Vitis-AI-Library/samples/dpu_task/yolov3/demo_yolov3.cpp的例子来进行修改，*.prototxt并不适用于这一API类别。
关于详细的API定义及定制化模型部署流程，可以参考ug1354 https://www.xilinx.com/html_docs/vitis_ai/1_3/fub1606771871673.html

LinErDeng, 2021-04-06:
那这块类别应该怎么设置呢？假设我只有一个类别，ids直接写1吗？
vector<int> coco_id_map_dict() {
  vector<int> category_ids;
  category_ids = {1,  2,  3,  4,  5,  6,  7,  8,  9,  10, 11, 13, 14, 15,
                  16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 27, 28, 31, 32,
                  33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 46, 47,
                  48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61,
                  62, 63, 64, 65, 67, 70, 72, 73, 74, 75, 76, 77, 78, 79,
                  80, 81, 82, 84, 85, 86, 87, 88, 89, 90};
  return category_ids;
}

fanz-xlnx, 2021-04-07:
不是的，你要找到你训得那一类对应的id是多少。
如果你用的是coco的数据集，那么可以参考这个https://github.com/tensorflow/models/blob/master/research/object_detection/data/mscoco_label_map.pbtxt

LinErDeng, 2021-04-09:
您好，无论我用哪个模型测./test_video_XXX ，都只有4.3FPS，会不会因为测试的时候屏幕上保留这样的警告：
** (test_video_yolov3:1023): CRITICAL **: 12:25:04.536: gst_v4l2_object_destroy: assertion 'v4l2object != NULL' failed

** (test_video_yolov3:1023): CRITICAL **: 12:25:05.013: gst_v4l2_object_destroy: assertion 'v4l2object != NULL' failed

(test_video_yolov3:1023): GStreamer-CRITICAL **: 12:25:10.202: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed

LinErDeng, 2021-04-09:
您好，无论我用哪个模型测./test_video_XXX ，都只有4.3FPS，会不会因为测试的时候屏幕上出现这样的警告：
** (test_video_yolov3:1023): CRITICAL **: 12:25:04.536: gst_v4l2_object_destroy: assertion 'v4l2object != NULL' failed

** (test_video_yolov3:1023): CRITICAL **: 12:25:05.013: gst_v4l2_object_destroy: assertion 'v4l2object != NULL' failed

(test_video_yolov3:1023): GStreamer-CRITICAL **: 12:25:10.202: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed

Premduth, 2021-08-18:
Please try this in

https://github.com/Xilinx/Vitis-AI-Tutorials

All ML tutorials moved there

closed 2021-08-23 by Premduth

charles-str, 2022-02-28:
> scripts/darknet_convert.sh

scripts/darknet_convert.sh在哪里？没有找到呢，谢谢

imrickysu, 2022-02-28:
Do you mean this file? https://github.com/Xilinx/Vitis-AI-Tutorials/blob/master/Design_Tutorials/07-yolov4-tutorial/scripts/darknet_convert.sh 

@charles-str , please ask Vitis-AI related questions in https://github.com/Xilinx/Vitis-AI-Tutorials/issues. It would be easier for other uses to find related issues.

charles-str, 2022-02-28:
thanks  so much.