Xilinx/Vitis-Tutorials #62: .dtsi file for ultra96 v2

JianmingTONG, 2021-04-03:
Hi,

I have a question regarding the Vitis_Platform_Creation:
```
https://github.com/Xilinx/Vitis-Tutorials/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md
```
What should I fill for the .dtsi file of ultra96 v2.

imrickysu, 2021-04-15:
For the reference [dtsi ](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/ref_files/step2_petalinux/system-user.dtsi)provided in the sample code, 

- bootargs, zocl and axi_intc_0 nodes can be kept same.
- sdhci1 is not needed.
- Other Ultra96 specific device tree changes need to refer to Ultra96 doc.

Premduth, 2021-08-18:
Closing under assumption it is resolved

closed 2021-08-18 by Premduth