Xilinx/Vitis-Tutorials #64: Accuracy drops to 0.098 on ZCU102 : Machine_Learning/Design_Tutorials/02-MNIST_classification_tf

nimaomidsajedi, 2021-04-12:
Hello. I am trying to deploy MNIST classification example on ZCU102 board but the accuracy drops to 0.098. Can you please consider this issue?

[image]

nhphuong91, 2021-04-28:
> I am using ZCU104 and the accuracy is 0.098 too, have you solved this problem?

I tried the 09-mnist_pyt-master and it works fine on the ZCU104. Didn't try the 02_mnist yet, but could you all test the 09_mnist?

xdzhang-ai, 2021-05-06:
> > I am using ZCU104 and the accuracy is 0.098 too, have you solved this problem?
> 
> I tried the 09-mnist_pyt-master and it works fine on the ZCU104. Didn't try the 02_mnist yet, but could you all test the 09_mnist?

thank you for your reply. I tried the 09-mnist_pyt-master, but the accuracy is still 0.098, did you use the pre-built dpu platform provided in introduction/module1 (https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2020.2-v1.3.0.img.gz)?

nhphuong91, 2021-05-07:
> thank you for your reply. I tried the 09-mnist_pyt-master, but the accuracy is still 0.098, did you use the pre-built dpu platform provided in introduction/module1 (https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2020.2-v1.3.0.img.gz)?

Yep, I use that one

lihengyi-ai, 2021-06-03:
I am using ZCU-102, and the accuracy decreases to 0.098 too,  is there someone has solved the problem?

giulio-naggi, 2021-06-03:
> 
> 
> I am using ZCU-102, and the accuracy decreases to 0.098 too, is there someone has solved the problem?

I had the same issue, but managed to resolve it by downloading the updated version of the [image file](https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu104-dpu-v2020.2-v1.3.1.img.gz)

lihengyi-ai, 2021-06-03:
> I had the same issue, but managed to resolve it by downloading the updated version of the [image file](https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu102-dpu-v2020.2-v1.3.1.img.gz)

Thank you very much, I will try it right now.

lihengyi-ai, 2021-06-03:
> Thank you very much, I will try it right now.

Although the problem is unknown,  it dose work to reload the image file~I just downloaded it yesterday.

Premduth, 2021-08-18:
Is this still open or resolved?

If unresoloved:

Please try this in

https://github.com/Xilinx/Vitis-AI-Tutorials

All ML tutorials moved there

closed 2021-08-23 by Premduth