Xilinx/Vitis-Tutorials #65: makefile and readme needs modification

vivianyin, 2021-04-15:
https://github.com/Xilinx/Vitis-Tutorials/blob/master/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial/Makefiles/Makefile

Line 63, SYSROOT_PATH needs to be changed with the path to xilinx-versal-common-v2020.2 at user's end. This should be added to Readme.

Line 247, --package.sd_dir $(PLATFORM_REPO_PATHS)/sw/versal/xrt
This line should be removed?

Line 248-249, the paths need to be changed with the path to xilinx-versal-common-v2020.2 at user's end. This should be added to Readme.

Premduth, 2021-08-18:
@ThomasXilinx  can you have someone look at this?

imrickysu, 2023-01-13:
Hi @SURUTHI1605 , I think this lenet issue has been solved. Please confirm.

SURUTHI1605, 2023-02-09:
I'm unable to access this link https://github.com/Xilinx/Vitis-Tutorials/blob/master/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial/Makefiles/Makefile

imrickysu, 2023-02-09:
In 2022.2 branch, the makefile path is now https://github.com/Xilinx/Vitis-Tutorials/blob/2022.2/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial/Makefile .

Please document the environment variables need to be set before running the makefile, e.g. `$COMMON_IMAGE_VERSAL` on top of the Makefile and README, and provide proper error messages if these variables are not set.

SURUTHI1605, 2023-05-07:
The environment variables need to be set was separately defined in sample_env_setup.sh script and the steps to reproduce the script was documented in README.

closed 2023-05-08 by imrickysu