Xilinx/Vitis-Tutorials #68: makefile need modification

yka2ki, 2021-05-05:
https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/example/zcu102/hw/Makefile

(**Original**)
package/sd_card.img: app.exe vadd.xclbin xrt.ini run_app.sh
    $(call ndef,ROOTFS)
    v++ -p -t ${TARGET} --config ../../src/zcu102.cfg ./vadd.xclbin **-o vadd.xclbin** \
        --package.out_dir package \
        --package.rootfs ${ROOTFS}/rootfs.ext4 \
        --package.sd_file vadd.xclbin \
        --package.sd_file ${ROOTFS}/Image \
        --package.sd_file xrt.ini \
        --**package.sd_file emconfig.json** \
        --package.sd_file app.exe \
        --package.sd_file run_app.sh

(**Modified**)
package/sd_card.img: app.exe vadd.xclbin xrt.ini run_app.sh
    $(call ndef,ROOTFS)
    v++ -p -t ${TARGET} --config ../../src/zcu102.cfg ./vadd.xclbin \
        --package.out_dir package \
        --package.rootfs ${ROOTFS}/rootfs.ext4 \
        --package.sd_file vadd.xclbin \
        --package.sd_file ${ROOTFS}/Image \
        --package.sd_file xrt.ini \
        --package.sd_file app.exe \
        --package.sd_file run_app.sh

Premduth, 2021-08-18:
@imrickysu  please look at this

imrickysu, 2021-08-21:
In Vitis 2021.1, the original Makefile generates error `ERROR: [v++ 60-2504] The input file and output file cannot be the same`. The Modified code can pass compilation, but it generates two xclbin files in sd_card directory: a.xclbin and vadd.xclbin.

Investigating how to reduce the redundancy.

Candice-Yin, 2022-01-06:
> https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/example/zcu102/hw/Makefile
>
> (**Original**) package/sd_card.img: app.exe vadd.xclbin xrt.ini run_app.sh $(call ndef,ROOTFS) [...]

Hi, I met the same problem, and then I modified it in your way. And I have another question, do you have any ideas? Thank you.

Candice-Yin, 2022-01-06:
Hi, I met the same question, and I modified it in your way. But then I have another question, do you have any ideas? Thank you.
[image]

imrickysu, 2022-01-06:
Hi @Candice-Yin , in your modified makefile, `package.out_dir` is being treated as a makefile target. Maybe you added an extra new line. Please check.

Candice-Yin, 2022-01-07:
I solved it. Thank you very much! @imrickysu 
During hardware emulation, I have an error about `./package/launch_hw_emu.sh`. Could you please have a look at it?
[image]

imrickysu, 2022-01-07:
Hi @Candice-Yin , I don't see a hard stop error in your messages.  Your Linux has boot up. There are some application level errors. Your Linux system should still work. If you hit enter, it should show you the input cursor.

Candice-Yin, 2022-01-07:
hi, it displayed “ Current directory: 0" when I continued typing.
[image]

imrickysu, 2022-01-07:
@Candice-Yin  It's a valid error message. Please add your xclbin file as an argument to app.exe.

Candice-Yin, 2022-01-07:
@imrickysu You helped me a lot, thx!

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx