Xilinx/Vitis-Tutorials #69: $(CXX) doesn't make sense for cross-compilation in Tutorials

vmayoral, 2021-05-07:
Cross-compilation of binaries at https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md is described as:

```bash
cd <Path to the cloned repo>/Getting_Started/Vitis/example/zcu102/sw_emu

$(CXX) -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I/usr/include/xrt -lOpenCL -lpthread -lrt -lstdc++
```

I don't really know what you mean with `$(CXX)`. Maybe it's a syntax I don't know (nor I can make it work). Do you mean `$(echo $CXX)`.

rwarmstr, 2021-05-07:
Like I mentioned in #70 this is an environment variable that's set when you source the cross-compilation environment. To avoid parallel threads let me close this one and we can continue the discussion there.

closed 2021-05-07 by rwarmstr

vmayoral, 2021-05-07:
@rwarmstr thanks for your comment. Please note the following in [here](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md):

[image]

Note 
```math
$(CCX) != ${CXX} 
```

I believe this should be changed.

rwarmstr, 2021-05-07:
Ah! Yes, you're right, I think the author may have misunderstood how this works on the shell vs. in a Makefile and I didn't notice it during review. Let me get it fixed and make a note to check for that in the future.