Xilinx/Vitis-Tutorials #70: aarch64-xilinx-linux-g++ missing, using aarch64-linux-gnu-g++ instead

vmayoral, 2021-05-07:
Connected to #69.

The instructions at https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md read as:
```bash
...
$(CXX) -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I/usr/include/xrt -lOpenCL -lpthread -lrt -lstdc++
...
```

cross-compilation wont' work this way since `CXX` contains the following in my environment with `Vitis 2020.2`:

```bash
echo $CCX
aarch64-xilinx-linux-g++ -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 --sysroot=/tools/Xilinx/Vitis/2020.2/data/emulation/qemu/unified_qemu_v5_0/sysroots/aarch64-xilinx-linux
```

`aarch64-xilinx-linux-g++` is nowhere to be found in my file system. Instead, I used `aarch64-linux-gnu-g++` to successfully cross-compile things.

randyh62, 2021-05-07:
I am not sure where this instruction or the $(CXX) variable came from. It seems to be a Makefile variable, but I could not find the `environment-setup-aarch64-xilinx-linux` script described in the line "source <path to the SDK>/environment-setup-aarch64-xilinx-linux" in the tutorial. The command, which you found, should be located as follows:

`$XILINX_VITIS/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++`

rwarmstr, 2021-05-07:
${CXX} is an environment variable that's set when you source the cross-compilation environment set up script. That's a necessary step when building on the command line.

It looks like you may not have done this before building and it's picking up the QEMU toolchain instead, which might accidentally work in some circumstances but can cause issues generally.

closed 2021-05-10 by vmayoral

dj-park, 2023-06-23:
For those of you still having similar issues... in my experience, when installing zynqmp command image,
if I install it with 
`./sdk.sh -y -d . -p`
and then cp it to /opt/platforms (for example), I had this issue.

I had to cp the directory to /opt/platforms/ first and then run `./sdk.sh -y -d . -p`.