Xilinx/Vitis-Tutorials #73: Xilinx Vitis vs Vitis HLS

m-kru, 2021-05-10:
After going through the tutorials for the `Xilinx Vitis` and `Vitis HLS` programs I still find vague:
1. What is the conceptual difference?
2. Why are there 2 distinct programs?
3. How do I know which one should I choose?

If I understand it correctly, `Xilinx Vitis` allows for developing both the software and the firmware (kernel) in the same time within single IDE, whereas `Vitis HLS` is only for the firmware (kernel) development. Please correct me if I am wrong.

`Vitis HLS` allows for `Vitis Kernel Flow`, however the firmware (kernel) can also be implemented in the `Xilinx Vitis`. Why would someone choose `Vitis HLS` with `Vitis Kernel Flow`, if the same can be done in `Xilinx Vitis`, plus it has the support also for the software side?

The only clear scenario I can see (although I am not 100 % sure), is that when someone wants to generated IP for the use with the `Vivado`, then choosing `Vitis HLS` with the `Vitis IP Flow` is the only way to go.

vmayoral, 2021-05-10:
@m-kru I also struggled making sense out of this, it's confusing and should be improved.
In case it helps, here's my understanding (though there are others here who know this much better):

> 1. What is the conceptual difference?

**Vitis** is the software platform, includes hooks and connections with all the rest software tools including HLS and Vivado.
**HLS** is a high-level synthesis tool that allows the creation of hardware kernels  in three languages:  C, C++, and OpenCL. **Vivado** is a design suite that includes tools for developing hardware kernels in RTL.

For more:
- [Vitis documentation](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/vitis_embedded_getstarted.html#rjm1602134444043)
- [Vitis HLS documentation](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/introductionvitishls.html#dvj1582158621847)

> 2. Why are there 2 distinct programs?

Well, there're four actually if you consider Vivado and Vitis AI. AFAIK there're two main reasons:
- Software modularity helps future workflows by helping developers "*select what they need*", and not more. 
- Historically HLS aspects were separated to meet the needs of different stakeholders

The following map helped me:

| Tool | Stakeholder | Aim |
|----|----------|--------------|
| Vivado | Hardware engineer | Develops RTL kernels |
| Vitis | Embedded engineer | Creates low level firmware, BSPs, boot sequences and integrates software and hardware efforts | 
| Vitis HLS | Software engineer | Develops C, C++ or OpenCL kernels |
| Vitis AI | Data scientist | Develops AI  constructs/kernels |

> 3. How do I know which one should I choose?

Depends on your use case and who are you. You might actually fit into various of the stakeholders groups above and thereby, coherently, you'll need to use more than one tool.

vmayoral, 2021-05-10:
> If Xilinx Vitis was only for integration then it would be simple, use Vitis HLS for kernel development and Xilinx Vitis for integration.

I think this piece of input is useful @m-kru, thanks.

> But now, for writing custom HLS kernels I can choose Xilinx Vitis or Vitis HLS, and after digging through multiple datasheets and tutorials does not even mention to bother which one should be used when, and why they provide non orthogonal tools.

I agree there's room for improvement.

Are your questions addressed @m-kru or are you still struggling to decide? If so, can you share of your setup and use case/aims (please be thorough if so) so that others more knowledgeable could advise?

randyh62, 2021-05-10:
It is a suite of tool that works together to build a system: 
1. The Vivado Design Suite does RTL synthesis and place and route to create a bitstream for the Programmable Logic (PL) region of a XIlinx device. It includes the IP integrator feature to do system level design linking multiple RTL IP. 
2. Vivado HLS _provided_ C/C++ entry into RTL design for Vivado, and creates IP that can be used in IP Integrator. It is legacy and replaced by Vitis HLS. 
3. Vitis HLS **provides** C/C++ entry into RTL design for Vivado, and also creates compiled XO objects (kernels) for Vitis. 
4. Vitis also includes the SW programming environment for embedded processors (PS), including OS and application development
5. Vitis uses Vitis HLS to compile the XO, and uses Vivado to link the system in IP Integrator and synthesize and place and route the bitstream to generate the XCLBIN. Vitis lets you: 
  - compile C/C++ into multiple XO objects (PL), and into host applications for an x86 or embedded processor (PS)
  - link those XO into a system, synthesize and place and route it in Vivado, and generate the XCLBIN (bitstream) for programming the hardware (PL), 
  - and connect the bitstream (PL) and host application (PS). 
6. Vitis also includes the AI Engine development environment for using the AI Engine array (AIE) in Versal devices, integrating PS-PL-AIE regions into a single system.

closed 2021-05-10 by m-kru