Xilinx/Vitis-Tutorials #74: Unable to reprogram FPGA and run NEW kernels in combination with XRT, affects both non-DFX and DFX base platforms

vmayoral, 2021-05-10:
#### Context

This ticket's part of a debugging effort and connected to past issues including #69 #70 #71 and #72.

While developing kernels it's common to test (also in `hw` target) incremental variants of it. So far, unfortunately, I haven't been able to do so. **Only the first kernel produced** (the first time the sd card is introduced is the hardware) **remains**. I haven't been able to successfully change this. The current issues has been validated in the following hardware and base platform combinations:

| Hardware | Base platform |
|-----------|---------------|
| ZCU102 | xilinx_zcu102_base_202020_1 |
| ZCU102 | xilinx_zcu102_base_dfx_202020_1 |
| ZCU104 | xilinx_zcu104_base_202020_1 |

A simple way to reproduce this is to build and test variants  (silly ones, simple modifications of the kernel) of https://github.com/Xilinx/Vitis-Tutorials/tree/master/Getting_Started/Vitis. An exemplary flow to reproduce this issue is as follows:

1. Follow https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md and build the sd_card.img
2. Flash raw image into SD card and boot
3. Run `app.exe` obtaining results coherent to the `vadd.xclbin` kernel (whatever they are)
4. Modify `vadd.cpp` ([source code](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/example/src/vadd.cpp)) introduce a small (or big, up to you) change
5. Rebuild kernel with Vitis `v++` compiler
6. Scp kernel to hardware as `vadd.xclbin.new`
7. Run `app.exe vadd.xclbin.new` 

The expected observation is that result remains fixed to **3.** and not related to the changes introduced in **4.**, **5.** and **6.**. A past issue in XRT identified a similar behavior https://github.com/Xilinx/XRT/issues/2747.

#### Results so far

**NOTE**: *I'm using Vitis suite 2020.2 on top of Ubuntu 20.04.*
**NOTE 2**: *Ignore paths, they're specific to my development machines.*

<details><summary>xilinx_zcu102_base_202020_1</summary>

Regardless of what I do the result's always the same which corresponds with the first kernel I flashed which modified https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/example/src/vadd.cpp#L15 to:

```math
out = in1 + 2 * in2
```

```bash
cd /home/xilinx/Vitis-Tutorials/Getting_Started/Vitis/example/zcu102/hw; rm -r package/ _x/ vadd.x* app.exe a.xclbin v++*

# configure env.
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export PLATFORM_REPO_PATHS=/home/xilinx/xilinx_zcu102_base_202020_1
export ROOTFS=/home/xilinx/rootfs
source /home/xilinx/rootfs/sdk/environment-setup-aarch64-xilinx-linux

# build
${CXX} -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I/usr/include/xrt -lOpenCL -lpthread -lrt -lstdc++
v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo
v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin
v++ -p -t hw --config ../../src/zcu102.cfg ./vadd.xclbin --package.out_dir package --package.rootfs ${ROOTFS}/rootfs.ext4 --package.sd_file ${ROOTFS}/Image --package.sd_file xrt.ini --package.sd_file app.exe --package.sd_file vadd.xclbin --package.sd_file run_app.sh

# test acceleration
cd /media/sd-mmcblk0p1
export XILINX_XRT=/usr
./app.exe vadd.xclbin

root@zynqmp-common-2020_2:/media/sd-mmcblk0p1# ./app.exe vadd.xclbin
[  265.853107] [drm] Pid 1221 opened device
[  265.857065] [drm] Pid 1221 closed device
INFO: Found Xilinx Platform
[  265.863503] [drm] Pid 1221 opened device
[  265.869667] [drm] Pid 1221 closed device
[  265.873637] [drm] Pid 1221 opened device
INFO: Loading 'vadd.xclbin'
[  265.962660] [drm] zocl_xclbin_read_axlf The XCLBIN already loaded
[  265.962676] [drm] zocl_xclbin_read_axlf c82ababc-a138-4cb2-bfd8-595f8a4cd9b3 ret: 0
[  265.971245] [drm] bitstream c82ababc-a138-4cb2-bfd8-595f8a4cd9b3 locked, ref=1
[  265.978917] [drm] Reconfiguration not supported
[  265.990677] [drm] bitstream c82ababc-a138-4cb2-bfd8-595f8a4cd9b3 unlocked, ref=0
[... 256 lines cut ...]
[  456.320444] rcu: 	1-...0: (1 GPs behind) idle=e82/1/0x4000000000000000 softirq=2437/2438 fqs=2596
[  456.329391] 	(detected by 3, t=5257 jiffies, g=8457, q=24)
[  456.334858] Task dump for CPU 0:
[  456.338071] swapper/0       R  running task        0     0      0 0x0000000a
[  456.345109] Call trace:
[  456.347548]  __switch_to+0x1c4/0x288
[  456.351108]  0xffff8000110d98a8
[  456.354232] Task dump for CPU 1:
[  456.357443] sh              R  running task        0  1257   1253 0x00000022
[  456.364482] Call trace:
[  456.366915]  __switch_to+0x1c4/0x288
[  456.370472]  0xfffffe0021cd95c0

(board hangs in here)
```

</details>

#### Questions
- I was surprised that this (by "this" I mean the development flows explained above in **Context**) couldn't work in non-DFX platforms. Is this expected to work in non-DFX platforms? If not, what's the proposal for reprogramming the FPGA with XRT flows?
- I was very surprised that I couldn't get this flow working even with DFX base platform. Can you confirm this *should* work with DFX platforms?
- Since none of the above worked and the [proposed fix in a past issue](https://github.com/Xilinx/XRT/issues/2747#issuecomment-583664143) only led to more issues, how can I manually reprogram FPGAs for using them with XRT?

I'm currently blocked with this (and have been blocked by it for more than a week already) so It'd be great if someone could help with it or suggest and alternative route. Anything for `ZCU102` or `ZCU104` would do. 
Thanks!

vmayoral, 2021-05-10:
It'd be awesome if I could get your thoughts on this @rwarmstr, @randyh62 or someone else.

vmayoral, 2021-05-10:
All right, after a few good hints (thanks @randyh62) and insisting on power-booting hardware platforms I conclude the following for non-DFX platforms:

1. Rebuilding an image completely (including rebuilting the kernel, etc) as described in the Vitis-Tutorials **DOES** get the kernel to reprogram the FPGA successfully. This requires:
  - modify kernel
  - `v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo`
  - `v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin`
  - `v++ -p -t hw --config ../../src/zcu102.cfg ./vadd.xclbin --package.out_dir package --package.rootfs ${ROOTFS}/rootfs.ext4 --package.sd_file ${ROOTFS}/Image --package.sd_file xrt.ini --package.sd_file app.exe --package.sd_file vadd.xclbin --package.sd_file run_app.sh`
  - copy image into into sd, boot
  - launch kernel
2. Rebuilding the kernel, copying it, rebooting and launching again **doesn't** work, picks up older kernel:
  - modify kernel
  - `v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo`
  - `v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin`
  - scp kernel file `vadd.xclbin` into running image
  - reboot
  - launch kernel
3. Rebuilding the kernel, copying it, launching it, rebooting and launching again **doesn't** work, picks up older kernel:
  - modify kernel
  - `v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo`
  - `v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin`
  - scp kernel file `vadd.xclbin` into running image
  - launch kernel
  - reboot
  - launch kernel
4. Rebuilding the kernel, copying it, shuting down, booting up and launching again **doesn't** work, picks up older kernel:
  - modify kernel
  - `v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo`
  - `v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin`
  - scp kernel file `vadd.xclbin` into running image
  - launch kernel
  - reboot
  - launch kernel

Summarizing:

| Approach | Description | Result|
|-----------|-------------|-------|
| 1 |  recreating whole sd image | **works** |
| 2 | scp-ing new kernel, rebooting, launch | doesn't |
| 3 | scp-ing new kernel, launching, rebooting, launch | doesn't |
| 4 | scp-ing new kernel, power-off, booting, launch | doesn't |

Why's only option **1.** working? Can someone explain what are the processes behind reprogramming the FPGA in non-DFX platforms? Do I need to recreate the whole raw image (for the sd card) with `v++ -p` every single time? where's this being reflected files-wise (i.e. what's changing in the file system for the FPGA to be reprogrammed?).

vmayoral, 2021-05-11:
All right, further testing things out (and through good feedback provided by Randy, Wes, Terry and others) I fount out that for non-DFX base platforms what matters isn't really the `*.xclbin` file since this seems to be ignored completely. Instead, the corresponding FPGA is programmed by the bitstream contained into the `BOOT.BIN` file.

This then leads to the following fifth approach:
5. Rebuilding the kernel, packaging it for generation of BOOT.BIN, copying BOOT.BIN to boot partition,rebooting up and launching again **does** work:
   - modify kernel
   - `v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo`
   - `v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin`
   - `v++ -p -t hw --config ../../src/zcu102.cfg ./vadd.xclbin --package.out_dir package --package.no_image`
   - scp  `BOOT.BIN` into boot partition
   - reboot
   - launch kernel

Summarizing:

| Approach | Description | Result|
|-----------|-------------|-------|
| 1 |  recreating whole sd image | **works** |
| 2 | scp-ing new kernel, rebooting, launch | doesn't |
| 3 | scp-ing new kernel, launching, rebooting, launch | doesn't |
| 4 | scp-ing new kernel, power-off, booting, launch | doesn't |
| 5 | scp-ing BOOT.bin, reboot, launch | **works** |

This addresses questions above 1 and 3, but 2 remains:

- ~~I was surprised that this (by "this" I mean the development flows explained above in **Context**) couldn't work in non-DFX platforms. Is this expected to work in non-DFX platforms? If not, what's the proposal for reprogramming the FPGA with XRT flows?~~
- I was very surprised that I couldn't get this flow working even with DFX base platform. Can you confirm this *should* work with DFX platforms?
- ~~Since none of the above worked and the [proposed fix in a past issue](https://github.com/Xilinx/XRT/issues/2747#issuecomment-583664143) only led to more issues, how can I manually reprogram FPGAs for using them with XRT?~~

Premduth, 2021-08-18:
@imrickysu  can you look at this?

dj-park, 2022-04-29:
In fact, I am experiencing the exact same issue.
For instance,
app_0.exe vadd_0.xclbin => test passed
reboot
app_1.exe vadd_1.xclbin => test passed

But if I don't reboot, for example,
app_0.exe vadd_0.xclbin => test passed
app_1.exe vadd_1.xclbin => hangs

mruizglz, 2022-06-09:
I have exactly the same behaviour than @dj-park. I am using a ZCU102 (DFX) and 2021.1 tools. Is there any solution to this problem?

imrickysu, 2022-06-09:
Hi @dj-park , @mruizglz , thanks for your feedback. I'll try to reproduce my end and follow up.

xerpi, 2023-04-08:
If I'm not mistaken, when you target the real hardware (' t hw'), a `boot.bin` file that contains the bitstream is created, which then you can copy to the boot partition on the SD to load that bitstream at boot.
Which would be equivalent when targeting the hardware emulator (`-t hw_emu`)?

shantirao, 2024-02-19:
Still the case in 2023.2. Is there a way to put the PL in the xclbin file instead of the boot.bin file? Or maybe the FPGA has to boot before petalinux does, so there's no way to hot-swap the PL?