Xilinx/Vitis-Tutorials #79: YOLOv4 KeyError: 'up_sampling2d/mul'

vjsrinivas, 2021-05-19:
Hello,
I've been trying to follow the YOLOv4 tutorial for Vitis AI 1.3, and I am stuck with the compilation stage where I get the following error: 

```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=['1,416,416,3'], layout='NHWC', model_files=['./quantize/quantize_eval_model.pb'], model_type='tensorflow', out_filename='./compiled/yolov4_org.xmodel', proto=None)
in_shapes: [[1, 416, 416, 3]]
[INFO] tensorflow model: quantize/quantize_eval_model.pb
[INFO] parse raw model     :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████| 526/526 [00:00<00:00, 25868.00it/s]             
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 284, in __create_xmodel_from_tf1
    cls.__specialcase_replace_elemadd_with_biasadd(xmodel)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 3406, in __specialcase_replace_elemadd_with_biasadd
    xmodel.topsort()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/ir/xmodel.py", line 190, in topsort
    indegree_xnode[cname] -= 1
KeyError: 'up_sampling2d/mul'
```

When I run with Vitis AI 1.2, it compiles fine but outputs the older elf format, not xmodel. Is there something I am missing from my compile script? Also, I converted the Darknet weights to PB with the YOLOv3 model-set project that was recommended in the tutorial, and both the normal and quantized weights run fine in Vitis. 

``` bash
TARGET=ZCU104
NET_NAME=yolov4

ARCH=/opt/vitis_ai/compiler/arch/DPUCZDX8G/${TARGET}/arch.json

vai_c_tensorflow --frozen_pb ./quantize/quantize_eval_model.pb \
		 --arch ${ARCH} \
		 --output_dir ./compiled \
		 --net_name ${NET_NAME} \
		 --options "{'mode':'normal', 'save_kernel':'', 'input_shape':'1,416,416,3'}"
```

Premduth, 2021-08-18:
Please try this in

https://github.com/Xilinx/Vitis-AI-Tutorials

All ML tutorials moved there

vjsrinivas, 2021-08-22:
For anyone else who encounters this error statement, I tried exporting with Vitis AI **1.4** and it was able to compile. I haven't tried to see if the xmodel runs correctly on my 104, but I will close this issue since the main problem is solved.

closed 2021-08-22 by vjsrinivas