Xilinx/Vitis-Tutorials #85: Partitioning Vitis AI SubGraphs on CPU/DPU - EnvironmentNotWritableError

nhphuong91, 2021-06-23:
Hi, I'm following _Partitioning Vitis AI SubGraphs on CPU/DPU_ tutorial but stuck at the [Install Packages and Patches on the Vitis AI Tools Container step](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Feature_Tutorials/pytorch-subgraphs#33-install-packages-and-patches-on-the-vitis-ai-tools-container). See screenshot below:
[image]

>(vitis-ai-pytorch) Vitis-AI /workspace/packages > conda install unilog-1.3.2-h7b12538_35.tar.bz2
>
>Downloading and Extracting Packages
######################################################################################################################## | 100%
>Preparing transaction: done
>Verifying transaction: failed
>
>EnvironmentNotWritableError: The current user does not have write permissions to the target environment.
  environment location: /opt/vitis_ai/conda/envs/vitis-ai-pytorch
 > uid: 1020
 > gid: 1023

Can anyone support me with it? Thanks in advanced!

Premduth, 2021-08-18:
@nhphuong91 can you try this in the new tutorials in https://github.com/Xilinx/Vitis-AI-Tutorials. If it is still not working there, open the issue there and close this one

nhphuong91, 2021-08-19:
@Premduth Thanks but I found that the _Partitioning Vitis AI SubGraphs on CPU/DPU_ tutorial no longer exist on that new repo

Premduth, 2021-08-19:
@nhphuong91  can you add this comment to the https://github.com/Xilinx/Vitis-AI-Tutorials? Someone there can comment on the status. Not sure why that would not be there anymore

Premduth, 2021-08-23:
Closing out this one to open in https://github.com/Xilinx/Vitis-AI-Tutorials so the right people can answer

closed 2021-08-23 by Premduth