Xilinx/Vitis-Tutorials #86: Missing .prototxt file after compilation and my md5sum.txt file doesn‘t have .prototxt

littlemww, 2021-07-03:
I'm trying to run test_accuracy_yolov3_voc on the ZCU104 but it aborts because of a missing .prototxt file.

 ./test_accuracy_yolov3_voc 2007_test_gt.txt yolov4_person_pred.txt

[libprotobuf ERROR google/protobuf/text_format.cc:324] Error parsing text-format vitis.ai.proto.DpuModelParamList: 1:5: Message type "vitis.ai.proto.DpuModelParamList" has no field named "name".

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0610 13:48:07.609874  1187 configurable_dpu_task_imp.cpp:140] Check failed: ok cannot parse config file. config_file=/usr/share/vitis_ai_library/models/yolov4_person/yolov4_person.prototxt
*** Check failure stack trace: ***

But I have copy the original yolov4_person.prototxt (Generated when I run the script ：bash scripts/darknet_convert.sh )to the board. 
[image]

when I open the md5sum.txt file generated at compile time(I compiled according to the yolov4 tutorial, and there were no errors when compiling) I found that there was only one line inside：
[image]
but I see other md5sum.txt in the model are as follows:
[image]

So Is it because of the lack of information of prototxt in this file？If so, how to generate the correct md5sum.txt file at compile time？

Thanks in advance

Premduth, 2021-08-18:
@allyzhou should this be moved to the Vitis-AI repo?

allyzhou, 2021-08-23:
> @allyzhou should this be moved to the Vitis-AI repo?

Yes, this should be transferred to https://github.com/Xilinx/Vitis-AI-Tutorials, @ryanvergel can you help transfer issue to that repo. Thanks.

Premduth, 2021-08-23:
@littlemww Can you look at the new repo: https://github.com/Xilinx/Vitis-AI-Tutorials and if it is still there, open the issue there? I am going to close this issue in this repo for now..

closed 2021-08-23 by Premduth