Xilinx/Vitis-Tutorials #87: DPU example on ZCU104 with 2021.1 tools

ksstms, 2021-07-20:
Hi! I'm having some issues with the [ZCU104 Vitis embedded platform tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104). I tried it with 2020.2 and 2021.1 version tools. I used the provided makefiles in both cases to create the platform, then did the last step in the Vitis GUI to build the DPU example project. The 2020.2 version worked without any issues. However, when doing the same in 2021.1, I get the following error when building the DPU example project:

```
../src/app/samples/src/resnet50.cpp:25:10: fatal error: opencv2/core.hpp: No such file or directory
```

There are some additional warnings with 2021.1 as well. I have shortened the absolute paths in the first two.
```
Invalid project path: Include path not found (<...>/02-Edge-AI-ZCU104/ref_files/step3_pfm/sw_comp/sysroots/cortexa72-cortexa53-xilinx-linux/usr/include/xrt/).
Invalid project path: Include path not found (<...>/02-Edge-AI-ZCU104/ref_files/vitis-workspace/dpu_trd/Emulation-SW/dpu_trd/src/app/samples/include).
Invalid project path: Include path not found (dpu_trd/Emulation-SW/dpu_trd/src/app/samples/include).
```

The `/usr/include/xrt` directory really is missing from the rootfs. I have checked the PetaLinux project, and it was not enabled. I manually followed step 2 of the tutorial to add the packages, but the build failed with the following error:
```
ERROR: Nothing RPROVIDES 'packagegroup-petalinux-xrt' (but <...>/02-Edge-AI-ZCU104/ref_files/step2_petalinux/build/petalinux/components/yocto/layers/meta-petalinux/recipes-core/images/petalinux-image-minimal.bb RDEPENDS on or otherwise requires it)
```

The 2nd and 3rd warnings are the same _(I guess)_. The build configuration is set to Hardware, so I have no idea why it's checking the `Emulation-SW` directory. Setting the library search path to an absolute path (instead of the workspace path shown in the tutorial) makes no difference.

Premduth, 2021-08-18:
@imrickysu  can you look into this?

imrickysu, 2021-08-21:
@ksstms Could you take a look at newly released tutorial for [2021.1](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md)? It should have addressed these issues because PetaLinux 2021.1 has some updates.

`packagegroup-petalinux-xrt` is not needed in 2021.1. Instead, please use `xrt` package. Contents in `packagegroup-petalinux-xrt` were added to the dependencies of `xrt`. PetaLinux will add them automatically.

ksstms, 2021-08-23:
Thank you, that solves the XRT issue.

Unfortunately the `packagegroup-petalinux-vitisai` will still cause a build failure. I see it in [step 4](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step4.md#create-the-design) that it's not currently possible to build the Vitis AI deps in the PetaLinux project. Is it possible to merge the downloaded sysroots content into my own one?

imrickysu, 2021-08-24:
Technically you can merge the two sysroots. But it's not a tested flow. 
VAI 1.4 PetaLinux recipe will be released soon. After it's released, you would be able to generate one single sysroot from a PetaLinux project.

imrickysu, 2021-08-24:
Update: vitis-ai recipes have been published. 
https://github.com/Xilinx/meta-vitis-ai 

Now you can add this layer in PetaLinux and build a final sysroot.
Refer to UG1144 for how to add custom layers. Search for key word "add layer".

closed 2021-08-31 by ksstms