Xilinx/Vitis-Tutorials #91: RTL Simulation runs fail for VADD RTL Tutorial

lmaxeniro, 2021-08-05:
Hi when I follow up the tutorial here for running the RTL simulation: [instructions:](https://github.com/Xilinx/Vitis-Tutorials/blob/2020.1/Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow/vivado_ip.md)

I get below error:
`ERROR: [VRFC 10-2989] 'axi_vip_pkg' is not declared [/home/xeniro/vivado_prj/rtl_kernel/rtl_kernel.srcs/sources_1/imports/src/testbench/Vadd_A_B_tb.sv:6]
ERROR: [VRFC 10-2989] 'slv_m00_axi_vip_pkg' is not declared [/home/xeniro/vivado_prj/rtl_kernel/rtl_kernel.srcs/sources_1/imports/src/testbench/Vadd_A_B_tb.sv:7]
ERROR: [VRFC 10-2989] 'slv_m01_axi_vip_pkg' is not declared [/home/xeniro/vivado_prj/rtl_kernel/rtl_kernel.srcs/sources_1/imports/src/testbench/Vadd_A_B_tb.sv:8]
ERROR: [VRFC 10-2989] 'control_Vadd_A_B_vip_pkg' is not declared [/home/xeniro/vivado_prj/rtl_kernel/rtl_kernel.srcs/sources_1/imports/src/testbench/Vadd_A_B_tb.sv:9]
INFO: [VRFC 10-311] analyzing module Vadd_A_B_tb
ERROR: [VRFC 10-2865] module 'Vadd_A_B_tb' ignored due to previous errors [/home/xeniro/vivado_prj/rtl_kernel/rtl_kernel.srcs/sources_1/imports/src/testbench/Vadd_A_B`

it looks in the testbench file these module can not be found, some source files are missing?
`import axi_vip_pkg::*;
import slv_m00_axi_vip_pkg::*;
import slv_m01_axi_vip_pkg::*;
import control_Vadd_A_B_vip_pkg::*;`

randyh62, 2021-08-05:
In your prior post you pointed to the 2020.2 docs but said you were using the 2020.1 release. The tutorial is specific to the release. Was this the problem you identified and resolved? Which tutorial files are you working with? 

In your last issue, you have the following comment: "OK I find there are some extra steps that are need, in the application note just alongside the repo.. you really should update the documentation." Can you provide the URL to the application note you are referring to?

lmaxeniro, 2021-08-06:
The previous issue was solved by following this guide [https://github.com/Xilinx/Vitis-Tutorials/blob/2020.1/Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow/package_ip.md](url)
So I close that issue.

For this issue, I am following the guide of 2020.1 as well : [https://github.com/Xilinx/Vitis-Tutorials/blob/2020.1/Hardware_Accelerators/Feature_Tutorials/01-rtl_kernel_workflow/vivado_ip.md](url)

AnusheelXilinx, 2024-12-23:
Please test the latest tutorial steps and let us know if you still this issue while building the design. 
We do not see this issue in the internal testing. 

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx