Xilinx/Vitis-Tutorials #93: Tutorial disappeared while trying to follow (with 2020.1 version)

asantiago012, 2021-08-09:
The tutorial "Vitis-Tutorials/Hardware_Accelerators/Introduction"  has disappeared while i was trying to follow all the steps for an university project.

I was attempting to check the acceleration potential through a simple project and it was the optimal one, because it cheked all the theorical concepts from beggining to end for a simple application (showing execution times). 

Now that it is not available i would like to know why it has been deleted and if there is any more optimal and updated projects for my situation.

The objectives of my university project are:
- Check, analyze and study acceleration methodology
- Use and understand Vitis IDE
- Do a final comparation between different solutions for same algorithm in different languages executed in CPU (C, Python, Java), and the one designed and run in Vitis (emulation) and Alveo (execution).

This is a section of the project so the implementation must not be too extended (15-25 hours max.)

I have seen the Bloom Filter tutorial, but i don`t know if it fits better in terms of concepts and time.

Also, the only card available to test is an Alveo U50, and all the tutorials seem to be prepared for the U200 one.

Thanks you for your attention and i hope to solve this problem!

randyh62, 2021-08-09:
This must be what you were following? 
https://github.com/Xilinx/Vitis-Tutorials/tree/2020.2/Hardware_Accelerators/Introduction

AdriiDev, 2021-08-09:
Yes It is! Thank you! I was not able to get to that URL, appeared 404 error.

I have been checking in more detail and maybe both projects could be followed for my study.

My main question about new tutorials is that if they can be ported to the Alveo U50, and if it a diffucult task. 

Seems to be possible but in a hard way, quoting Bloom tutorial text: 

> TIP: This tutorial takes approximately two hours to complete.

> The labs in this tutorial use:
> - BASH Linux shell commands.
> - 2021.1 Vitis core development kit release and the xilinx_u200_gen3x16_xdma_1_202110_1 platform. If necessary, it can be easily ported to other versions and platforms.

> This tutorial guides you to run the designed accelerator on the FPGA; therefore, the expectation is that you have an Xilinx® Alveo™ U200 Data Center accelerator card set up to run this tutorial. Because it can take several (six or seven) hours to generate the multiple xclbin files needed to run the accelerator, pregenerated xclbin files are provided for the U200 card. To use these pregenerated files, when building the hardware kernel or running the accelerator on hardware, you need to add the SOLUTION=1 argument.

Is there any tutorial to port this kind of projects? The problem is if i have to compile and generate xclbin files each time i have to run the code (6-7h per proccess), it can take a long time.

randyh62, 2021-08-09:
Generally the tutorial can be run using a different platform pretty simply. If you look at the bloo filter tutorial, in the Makefile at the following URL: https://github.com/Xilinx/Vitis-Tutorials/blob/2020.2/Hardware_Accelerators/Design_Tutorials/02-bloom/makefile/Makefile

You will see line 21 of the Makefile has the following:
PLATFORM :=xilinx_u200_xdma_201830_2

You can edit this line, or simply specify the PLATFORM as part of the "make" command, substituting in the platform of your choice:
make build PLATFORM=xilinx_u50_gen3x16_xdma_201920_3 STEP=single_buffer PF=4 ITER=8

Most tutorials will run on multiple platforms, though some tutorials use a specific feature of a platform that is not available on other platforms, and those may fail to build or run. 

In general I would suggest you work with the hw_emu target. rather than the hw target, as it will build much more quickly. When you need to run it on hardware you can build it for the hw which will take several hours as you know. You can do this by changing the TARGET= in the makefile (line 2) or specify the TARGET= in the make command.

AdriiDev, 2021-08-09:
Thank you very much! This helps me a lot!! Now im ready to go on. 
I will give it a try and check both tutorials to set a base of acceleration knowledge.

closed 2021-08-09 by asantiago012