Xilinx/Vitis-Tutorials #95: Failure while emulating Bloom Tutorial in an Alveo U50 platform

asantiago012, 2021-08-18:
Hi!

I have been trying to follow  [Bloom Filter Example](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Hardware_Accelerators/Design_Tutorials/02-bloom) and I have been experimenting some problems. First of all I had to modify some files:

`makefile/common.mk` (changed DDR to HBM):
```
./$(BUILDDIR)/runOnfpga_$(TARGET).xo: $(SRCDIR)/hls_stream_utils.h $(HOST_SRC_FPGA)
	v++ -c -g -t $(TARGET) -R 1 -k runOnfpga \
		--platform $(PLATFORM) \
        --profile.data all:all:all \
        --profile.stall all:all:all \
		--profile.trace_memory HBM[3] \
		--save-temps \
		--temp_dir ./$(BUILDDIR)/temp_dir \
		--report_dir ./$(BUILDDIR)/report_dir \
		--log_dir ./$(BUILDDIR)/log_dir \
		-I$(SRCDIR) \
		-DPARALLELISATION=$(PF) \
		$(HOST_SRC_FPGA) \
		-o ./$(BUILDDIR)/runOnfpga_$(TARGET).xo

ifeq ($(STEP),multiDDR)
./$(BUILDDIR)/runOnfpga_$(TARGET).xclbin: ./$(BUILDDIR)/runOnfpga_$(TARGET).xo
	v++ -l -g -t $(TARGET) -R 1 \
		--platform $(PLATFORM) \
        --profile.data all:all:all \
        --profile.stall all:all:all \
		--profile.trace_memory HBM[3] \
		--save-temps \
		--temp_dir ./$(BUILDDIR)/temp_dir \
		--report_dir ./$(BUILDDIR)/report_dir \
		--log_dir ./$(BUILDDIR)/log_dir \
		--config $(CURRENT_DIR)/connectivity.cfg \
		-I$(SRCDIR) \
		./$(BUILDDIR)/runOnfpga_$(TARGET).xo \
		-o ./$(BUILDDIR)/runOnfpga_$(TARGET).xclbin

else

./$(BUILDDIR)/runOnfpga_$(TARGET).xclbin: ./$(BUILDDIR)/runOnfpga_$(TARGET).xo
	v++ -l -g -t $(TARGET) -R 1 \
		--platform $(PLATFORM) \
        --profile.data all:all:all \
	--profile.stall all:all:all \
		--profile.trace_memory HBM[3] \
		--save-temps \
		--temp_dir ./$(BUILDDIR)/temp_dir \
		--report_dir ./$(BUILDDIR)/report_dir \
		--log_dir ./$(BUILDDIR)/log_dir \
		-I$(SRCDIR) \
		./$(BUILDDIR)/runOnfpga_$(TARGET).xo \
		-o ./$(BUILDDIR)/runOnfpga_$(TARGET).xclbin

endif
```

`makefile/Makefile` (changed TARGET and PLATFORM):
```
TARGET := hw_emu
PLATFORM :=xilinx_u50_gen3x16_xdma_201920_3
```

I have not get to multiDDR section but I modified also connectivitu file for future.
`makefile/connectivity.cfg`:
```
[connectivity]
sp=runOnfpga_1.input_words:HBM[0:31] 
```

I had to solve those problems and some else for emulation and running purposes.

Now I am stucked on [Data Movement Between the Host and Kernel](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Hardware_Accelerators/Design_Tutorials/02-bloom/5_data-movement.md) because my emulation output is not really matching what tutorial expects to.

While tutorial gets a clean run:
[image]

I get a really dirty one (seems to be mixing concept from several sections of the tutorial or just failing hard):
[image]
This is the output for a `make run PLATFORM=xilinx_u50_gen3x16_xdma_201920_3 STEP=generic_buffer PF=8 ITER=8 TARGET=hw_emu` command. Also, beeing hw_emu, i get times in seconds because of execution time and not in ms of emulation time. Is there a solution for this?

I can understand that there is a huge difference between Alveo U200 and U50 because of DDR-HBM and also because of hw-hw_emu targets but i think that It should not be this much.

I need some help because I need to understand the methodology and get the tutorial complete. By now this is confusing and I can`t get so much conclusions.

Thank you and I am waiting for the response with great interest!

asantiago012, 2021-08-18:
UPDATE:

As expected, multiDDR behavior is similar:
[image]

Premduth, 2021-08-18:
@ThomasXilinx  can you have someone in your team look into this?

ravicho, 2021-08-24:
Hi Adrian,

When you run the hardware emulation, your application is not running on the card, and this mode uses behavior models to mimic the hardware. These are not precisely the same performance as that of when the application is run on the hardware. The advantage here is that you create a functional representation of the whole system, RTL being cycle-accurate. For example, you are not writing from the host to the FPGA device memory in hardware emulation but using a behavior model of the memory. The delay in migrating the data to the memory model will be different (much less)  in emulation vs. Hw. In the host code, each for loop iteration that enqueues the kernel is independent. A vector of events is created to coordinate migrating data from the host, enqueues the kernel, and migrate data back to the host. Due to this, the host can schedule several enqueues in advance but will execute one by one when the kernel is available to process the following data set.  I have attached the snapshot here to show you the dependencies in the timeline trace. You can see that the kennel en queue has a dependency on writing the buffer to the device memory. In hardware emulation, the delay modeled is much less than the actual hardware. So the host can schedule more kennel enqueues in advance.  
The tutorial explores how to minimize the gap of migrating buffer to the device memory and execute kernel for computing the data copied to the memory. You can see from the timeline that there are no gaps in Hw runs. You can also argue that even the hardware run may have a similar overlap like hw emulation if buffer transfer is much faster to the device memory than actual compute. I hope this clarifies some of your confusion. 

[image]

Regarding the time units, you can Right Click on the scale at the top and change the units as shown here. 

[image]

Let me know if we should schedule a meeting to go over these.

ravicho, 2021-08-30:
Hi Adrian,

Yes. Your understanding is correct. Let me know if you experience any further issues when you run this tutorial on the hw. 

Regards
Ravi

asantiago012, 2021-09-02:
Hello!

I am trying to execute it on Hardware and I am having so much troubles. It seems that it is not much easy to port to Alveo U50 because there are some errors like "std:bad_alloc" and others that i have been able to solve.

If it would be possible to shcedule a meeting, it would be amazing because i don´t know what else to modify in files and i don´t have much time to solve this situation. In the meeting I could be following the proccedure while we are talking about it.

The platform used is " xilinx_u50_gen3x4_xdma_base_2" and the Vitis/XRT version is "2020.1".

Thank you and I will be waiting for your info.

ravicho, 2021-09-02:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi Adrian,

Can you list all the changes you have implemented so far to make it work for u50 card?

I suspect that you are allocating a buffer more than the capacity of HBM 1 Pseudo channel, 256MB, resulting in bad allocation.  This is not a problem with DDR, as the capacity for each bank is 16G.

Please share what step you are running and the complete error for me to take a look.  I will schedule a meeting next week either way.

Regards
Ravi Chopra

ravicho, 2021-09-02:
Hi Adrian, Where should I send you an invite to the meeting?   Ravi

asantiago012, 2021-09-02:
Hi!

The step and output is:

```
>> make run STEP=single_buffer PF=4 TARGET=hw

..........
mkdir -p ./../build/single_buffer/kernel_4/hw
g++ -D__USE_XOPEN2K8 -D__USE_XOPEN2K8 \
	-I/opt/xilinx/xrt/include/ \
	-I./../reference_files \
	-O3 -Wall -fmessage-length=0 -std=c++11\
	./../reference_files/compute_score_host.cpp ./../reference_files/MurmurHash2.c ./../reference_files/xcl2.cpp ./../reference_files/main.cpp  ./../reference_files/run_single_buffer.cpp \
	-L/opt/xilinx/xrt/lib/ \
	-lxilinxopencl -lpthread -lrt \
	-o ./../build/single_buffer/kernel_4/hw/host
./../reference_files/compute_score_fpga_kernel.cpp is being used as source for Generating Kernel
cp xrt.ini ./../build/single_buffer/kernel_4/hw
************  Use Command Line to run application!  ************
cd ./../build/single_buffer/kernel_4/hw && ./host 100000  ;
Initializing data
Creating documents - total size : 1398.903 MBytes (349725824 words)
Creating profile weights

Loading runOnfpga_hw.xclbin
XRT build version: 2.8.726
Build hash: 7c93966ead2dec777b92bdc379893f22b5bd561e
Build date: 2020-11-11 20:29:19
Git branch: 2020.2
PID: 10294
UID: 1000
[Thu Sep  2 22:55:02 2021 GMT]
HOST: .........
EXE: ....../VitisTutorials/Hardware_Acceleration/Design_Tutorials/02-bloom/build/single_buffer/kernel_4/hw/host
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
Segmentation fault (core dumped)
Makefile:122: recipe for target 'run' failed
make: *** [run] Error 139
```

And the modifications, for each file:
- "makefile/common.mk" (tried with 3 and 31, no change):
```
./$(BUILDDIR)/runOnfpga_$(TARGET).xo: $(SRCDIR)/hls_stream_utils.h $(HOST_SRC_FPGA)
	v++ -c -g -t $(TARGET) -R 1 -k runOnfpga \
		--platform $(PLATFORM) \
		--profile_kernel data:all:all:all \
		--profile_kernel stall:all:all:all \
		--trace_memory HBM[31] \
		--save-temps \
		--temp_dir ./$(BUILDDIR)/temp_dir \
		--report_dir ./$(BUILDDIR)/report_dir \
		--log_dir ./$(BUILDDIR)/log_dir \
		-I$(SRCDIR) \
		-DPARALLELISATION=$(PF) \
		$(HOST_SRC_FPGA) \
		-o ./$(BUILDDIR)/runOnfpga_$(TARGET).xo

ifeq ($(STEP),multiDDR)
./$(BUILDDIR)/runOnfpga_$(TARGET).xclbin: ./$(BUILDDIR)/runOnfpga_$(TARGET).xo
	v++ -l -g -t $(TARGET) -R 1 \
		--platform $(PLATFORM) \
		--profile_kernel data:all:all:all \
		--profile_kernel stall:all:all:all \
		--trace_memory HBM[31] \
		--temp_dir ./$(BUILDDIR)/temp_dir \
		--report_dir ./$(BUILDDIR)/report_dir \
		--log_dir ./$(BUILDDIR)/log_dir \
[... 161 lines cut ...]

#####################################################################################################
###   For HW run, use 100,000 words for computation 
###   For Emulation run, use 100 words for computation 
#####################################################################################################

run:   build host
	cp xrt.ini $(BUILDDIR)
ifeq ($(TARGET), hw)
	@echo "************  Use Command Line to run application!  ************"
	cd $(BUILDDIR) && ./host 100000 $(ITER) ;
else
	@echo "Running $(TARGET) mode"; 
	emconfigutil --nd 1  --platform $(PLATFORM) --od $(BUILDDIR)
	cd $(BUILDDIR) && XCL_EMULATION_MODE=$(TARGET) ./host 100 $(ITER) ;
endif
```

These are my last files. I modified them several times for hw_emulation in other machine.

I have not modified source files on this version and maybe it is the problem. I didnt know there are those differences between models. The definitions and procedures are the same as the ones for U200 so if there is a difference in allocation sizes i have it all wrong. If it is neeeded to do a port process, please let me know how to do it. My job right now is to study the whole performance and potencial of the technology so I have not much knowledge about low-level coding.

Send the invite to the meeting to the email in my profile. If it is not updated please let me know.

Thanks and sorry for inconveninces.

asantiago012, 2021-09-03:
Just in case it helps, here is the size of the buffers printed while execution:
```
Creating buffers of size: 
     - bloom_filter: 65536 bytes
     - input_doc_words: 1398903296 bytes
     - output_inh_flags: 349725824 bytes
```
beeing obviously greater than the maximum of 256MB.

asantiago012, 2021-09-03:
New more info about buffer allocation:

- Highest working value for number of documents:
```
make run STEP=single_buffer PF=4 ITER=32 TARGET=hw NUM_DOCS=15374
mkdir -p ./../build/single_buffer/kernel_4/hw
g++ -D__USE_XOPEN2K8 -D__USE_XOPEN2K8 \
	-I/opt/xilinx/xrt/include/ \
	-I./../reference_files \
	-O3 -Wall -fmessage-length=0 -std=c++11\
	./../reference_files/compute_score_host.cpp ./../reference_files/MurmurHash2.c ./../reference_files/xcl2.cpp ./../reference_files/main.cpp  ./../reference_files/run_single_buffer.cpp \
	-L/opt/xilinx/xrt/lib/ \
	-lxilinxopencl -lpthread -lrt \
	-o ./../build/single_buffer/kernel_4/hw/host
./../reference_files/compute_score_fpga_kernel.cpp is being used as source for Generating Kernel
cp xrt.ini ./../build/single_buffer/kernel_4/hw
************  Use Command Line to run application!  ************
cd ./../build/single_buffer/kernel_4/hw && ./host 15374 32 ;
Initializing data
Creating documents - total size : 214.688 MBytes (53671936 words)
Creating profile weights

Loading runOnfpga_hw.xclbin

 Creating buffers of size: 
     - bloom_filter: 65536 bytes
     - input_doc_words: 214687744 bytes
     - output_inh_flags: 53671936 bytes
   
 Processing 214.688 MBytes of data
 Single_Buffer: Running with a single buffer of 214.688 MBytes for FPGA processing
--------------------------------------------------------------------
 Executed FPGA accelerated version  |   212.2754 ms   ( FPGA 145.258 ms )
 Executed Software-Only version     |    383.9105 ms
--------------------------------------------------------------------
 Verification: PASS
```

- Lowest not working value for number of documents:
```
make run STEP=single_buffer PF=4 ITER=32 TARGET=hw NUM_DOCS=15375
mkdir -p ./../build/single_buffer/kernel_4/hw
g++ -D__USE_XOPEN2K8 -D__USE_XOPEN2K8 \
	-I/opt/xilinx/xrt/include/ \
	-I./../reference_files \
	-O3 -Wall -fmessage-length=0 -std=c++11\
	./../reference_files/compute_score_host.cpp ./../reference_files/MurmurHash2.c ./../reference_files/xcl2.cpp ./../reference_files/main.cpp  ./../reference_files/run_single_buffer.cpp \
	-L/opt/xilinx/xrt/lib/ \
	-lxilinxopencl -lpthread -lrt \
	-o ./../build/single_buffer/kernel_4/hw/host
./../reference_files/compute_score_fpga_kernel.cpp is being used as source for Generating Kernel
cp xrt.ini ./../build/single_buffer/kernel_4/hw
************  Use Command Line to run application!  ************
cd ./../build/single_buffer/kernel_4/hw && ./host 15375 32 ;
Initializing data
Creating documents - total size : 214.704 MBytes (53676032 words)
Creating profile weights

Loading runOnfpga_hw.xclbin

 Creating buffers of size: 
     - bloom_filter: 65536 bytes
     - input_doc_words: 214704128 bytes
     - output_inh_flags: 53676032 bytes
 
XRT build version: 2.8.726
Build hash: 7c93966ead2dec777b92bdc379893f22b5bd561e
Build date: 2020-11-11 20:29:19
Git branch: 2020.2
PID: 9296
UID: 1000
[Fri Sep  3 06:02:58 2021 GMT]
HOST: ...................
EXE: ..................../VitisTutorials/Hardware_Acceleration/Design_Tutorials/02-bloom/build/single_buffer/kernel_4/hw/host
[XRT] ERROR: std::bad_alloc
Segmentation fault (core dumped)
Makefile:123: recipe for target 'run' failed
make: *** [run] Error 139
```

asantiago012, 2021-11-14:
I managed to solve the issue with help of @ravicho .

I executed succesfully all the steps of tutorial, except the last one, using several DDR banks. 

Here are all modifications on each file.

- makefile/Makefile:
    - Changed TARGET and PLATFORM
    - Added NUM_DOCS variable to set number of documents for tests

```
CURRENT_DIR = $(PWD)
TARGET := hw

#####################################################################################################
## Detemines how many batches of buffers are sent from host to kernel
#####################################################################################################
ITER   := 

#####################################################################################################
## PF sets how many words to process in parallel 
#####################################################################################################
PF     := 8
ifeq ($(STEP), kernel_4)
PF   := 4
endif
ifeq ($(STEP), kernel_16)
PF   := 16
endif

NUM_DOCS:=10000

#PLATFORM :=xilinx_u200_gen3x16_xdma_1_202110_1
#PLATFORM :=xilinx_u50_gen3x16_xdma_201920_3
PLATFORM :=xilinx_u50_gen3x4_xdma_2_202010_1

SRCDIR := ./../reference_files

# Default Settings here ..
HOST_SRC_CPP := $(SRCDIR)/compute_score_host.cpp
BUILDDIR := ./../build/$(STEP)/kernel_$(PF)/$(TARGET)

# Common Source Code for FPGA and Host
HOST_SRC_FPGA := $(SRCDIR)/compute_score_fpga_kernel.cpp
HOST_SRC_CPP := $(SRCDIR)/compute_score_host.cpp
HOST_SRC_CPP += $(SRCDIR)/MurmurHash2.c
HOST_SRC_CPP += $(SRCDIR)/xcl2.cpp
HOST_SRC_CPP += $(SRCDIR)/main.cpp 

#### For the following steps, don't rebuild the kernels as only Host is changing
ifeq ($(STEP), $(filter $(STEP), split_buffer , generic_buffer , sw_overlap ))
NOXCLBIN=1
endif

#####################################################################################################
##  User appropriate host file 
#####################################################################################################
HOST_SRC_RUN_CPP = $(SRCDIR)/run_single_buffer.cpp
ifeq ($(STEP),multiDDR)
HOST_SRC_RUN_CPP = $(SRCDIR)/run_sw_overlap_multiDDR.cpp
endif
ifeq ($(STEP),split_buffer)
HOST_SRC_RUN_CPP = $(SRCDIR)/run_split_buffer.cpp
endif
ifeq ($(STEP),generic_buffer)
HOST_SRC_RUN_CPP = $(SRCDIR)/run_generic_buffer.cpp
endif
ifeq ($(STEP),sw_overlap)
HOST_SRC_RUN_CPP = $(SRCDIR)/run_sw_overlap.cpp
endif
[... 190 lines cut ...]
```
[connectivity]
sp=runOnfpga_1.input_words:HBM[0:14] 
sp=runOnfpga_1.output_flags:HBM[15:29] 
sp=runOnfpga_1.bloom_filter:HBM[30]
```
 

//////////////////////////////////////////////////////////////////////////////////////////////////////////////

Also, following and adapting Choleskw Acceleration Tutorial I found an error on Makefile. I had to modify:

```
   100 $(BUILD_DIR)/$(HOST_EXE): $(HOST_SRC_CPP) $(HOST_SRC_H)
   101    mkdir -p $(BUILD_DIR)
   102    g++ $(CXXFLAGS) $(CXXLDFLAGS) -o $@ $(HOST_SRC_CPP)
```

to:

```
   100 $(BUILD_DIR)/$(HOST_EXE): $(HOST_SRC_CPP) $(HOST_SRC_H)
   101    mkdir -p $(BUILD_DIR)
   102    g++ $(CXXFLAGS) -o $@ $(HOST_SRC_CPP) $(CXXLDFLAGS)
```

closed 2021-11-14 by asantiago012