Xilinx/Vitis-Tutorials #99: Vitis Platform Creation Tutorial for ZCU104: Emulation Does Not Work

Hamptonjc, 2021-08-31:
Following: https://github.com/Xilinx/Vitis-Tutorials/tree/2020.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104#step-by-step-tutorial

During step 4, test 2 "Run Vector Addition Application", when running HW-emulation I get the following in the emulation console of the Vitis IDE:
```

Starting QEMU
 - Press <Ctrl-a h> for help 
Waiting for QEMU to start. qemu_port 8064
Qemu_pids 4450 4451
qemu-system-aarch64: -chardev socket,path=./qemu-rport-_pmu@0,server,id=pmu-apu-rp: info: QEMU waiting for connection on: disconnected:unix:./qemu-rport-_pmu@0,server

QEMU started. qemu_pid=4450.
Waiting for PMU to start. Qemu_pids 4454 4455
qemu-system-aarch64: -chardev socket,id=pl-rp,host=127.0.0.1,port=7724,server: info: QEMU waiting for connection on: disconnected:tcp:127.0.0.1:7724,server

PMC started. pmc_pid=4454
Starting PL simulation.Generating PLLauncher commandline
running PLL Launcher
qemu-system-aarch64: warning: hub 0 is not connected to host network

PMU Firmware 2020.2	Aug 31 2021   17:24:15
PMU_ROM Version: xpbr-v8.1.0-0
NOTICE:  ATF running on XCZUUNKN/QEMU v4/RTL0.0 at 0xfffea000
NOTICE:  BL31: v2.2(release):xlnx_rebase_v2.2_2020.3
NOTICE:  BL31: Built : 17:22:19, Aug 31 2021

U-Boot 2020.01 (Aug 31 2021 - 17:23:30 +0000)

Model: ZynqMP ZCU104 RevC
Board: Xilinx ZynqMP
DRAM:  2 GiB
PMUFW:	v1.1
EL Level:	EL2
Chip ID:	unknown
NAND:  0 MiB
MMC:   mmc@ff170000: 0
In:    serial@ff000000
Out:   serial@ff000000
Err:   serial@ff000000
Bootmode: JTAG_MODE
Reset reason:	
Net:   
ZYNQ GEM: ff0e0000, mdio bus ff0e0000, phyaddr 12, interface rgmii-id

Warning: ethernet@ff0e0000 using MAC address from DT
eth0: ethernet@ff0e0000
Hit any key to stop autoboot:  2  1  0 
JTAG: Trying to boot script at 0x20000000
## Executing script at 20000000
Wrong image format for "source" command
JTAG: SCRIPT FAILED: continuing...
starting USB...
Bus dwc3@fe200000: this is not a DesignWare USB3 DRD Core
failed to initialize core
Port not available.
BOOTP broadcast 1
DHCP client bound to address 10.0.2.15 (1 ms)
*** Warning: no boot file name; using '0A00020F.img'
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename '0A00020F.img'.
Load address: 0x8000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/01-00-0a-35-00-22-01
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/01-00-0a-35-00-22-01'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A00020F
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A00020F'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A00020
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A00020'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A0002
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A0002'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A000
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A000'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A00
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A00'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A0
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A0'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0A
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0A'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/0'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-zynqmp-zynqmp
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/default-arm-zynqmp-zynqmp'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-zynqmp
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/default-arm-zynqmp'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/default-arm'.
Load address: 0x10000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
```

Note: I am using branch 2020.2, matching my Xilinx tool versions.

Any ideas?

Premduth, 2021-08-31:
@imrickysu  can you take a look?

imrickysu, 2021-09-01:
@Hamptonjc This error behavior looks very like the u-boot can't find **boot.scr**.
Please make sure you put boot.scr into image directory when Preparing Files for Platform Packaging (step 6).

Hamptonjc, 2021-09-01:
@imrickysu I can confirm boot.scr is in the image directory:

[image]

Hamptonjc, 2021-09-01:
The full read out from the emulation console:

``` Current working dir /home/jhampton/board-dev/zcu104/zcu104_custom_pkg/vadd_system/Emulation-HW/package
Required emulation files like qemu_args exists

Starting QEMU
 - Press <Ctrl-a h> for help 
Waiting for QEMU to start. qemu_port 8118
Qemu_pids 20769 20770
qemu-system-aarch64: -chardev socket,path=./qemu-rport-_pmu@0,server,id=pmu-apu-rp: info: QEMU waiting for connection on: disconnected:unix:./qemu-rport-_pmu@0,server

QEMU started. qemu_pid=20769.
Waiting for PMU to start. Qemu_pids 20773 20774
PMC started. pmc_pid=20773
qemu-system-aarch64: -chardev socket,id=pl-rp,host=127.0.0.1,port=8876,server: info: QEMU waiting for connection on: disconnected:tcp:127.0.0.1:8876,server

Simulation Started. pl_pid=20779
qemu-system-aarch64: warning: hub 0 is not connected to host network

PMU Firmware 2020.2	Aug 31 2021   17:24:15
PMU_ROM Version: xpbr-v8.1.0-0
NOTICE:  ATF running on XCZUUNKN/QEMU v4/RTL0.0 at 0xfffea000
NOTICE:  BL31: v2.2(release):xlnx_rebase_v2.2_2020.3
NOTICE:  BL31: Built : 17:22:19, Aug 31 2021

U-Boot 2020.01 (Aug 31 2021 - 17:23:30 +0000)

Model: ZynqMP ZCU104 RevC
Board: Xilinx ZynqMP
DRAM:  2 GiB
PMUFW:	v1.1
EL Level:	EL2
Chip ID:	unknown
NAND:  0 MiB
MMC:   mmc@ff170000: 0
In:    serial@ff000000
Out:   serial@ff000000
Err:   serial@ff000000
Bootmode: JTAG_MODE
Reset reason:	
Net:   
ZYNQ GEM: ff0e0000, mdio bus ff0e0000, phyaddr 12, interface rgmii-id

Warning: ethernet@ff0e0000 using MAC address from DT
eth0: ethernet@ff0e0000
Hit any key to stop autoboot:  2  1  0 
JTAG: Trying to boot script at 0x20000000
## Executing script at 20000000
Wrong image format for "source" command
JTAG: SCRIPT FAILED: continuing...
starting USB...
Bus dwc3@fe200000: this is not a DesignWare USB3 DRD Core
failed to initialize core
Port not available.
BOOTP broadcast 1
DHCP client bound to address 10.0.2.15 (4 ms)
*** Warning: no boot file name; using '0A00020F.img'
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename '0A00020F.img'.
Load address: 0x8000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/01-00-0a-35-00-22-01
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'pxelinux.cfg/01-00-0a-35-00-22-01'.
[... 294 lines cut ...]
Config file not found
starting USB...
Bus dwc3@fe200000: this is not a DesignWare USB3 DRD Core
failed to initialize core
Port not available.
BOOTP broadcast 1
DHCP client bound to address 10.0.2.15 (0 ms)
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'boot.scr.uimg'.
Load address: 0x20000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
BOOTP broadcast 1
DHCP client bound to address 10.0.2.15 (0 ms)
Using ethernet@ff0e0000 device
TFTP from server 10.0.2.2; our IP address is 10.0.2.15
Filename 'boot.scr.uimg'.
Load address: 0x18000000
Loading: *
TFTP error: 'Access violation' (2)
Not retrying...
ZynqMP> 
```

imrickysu, 2022-08-01:
Hi @sven2314 Please check this issue and provide suggestions.

sven2314, 2022-08-23:
Hi, @Hamptonjc 
From my experience this error is caused by lacking of boot.scr.  Most importantly the boot.scr should be prepared in image directory and give it to Vitis before generating the platform.  If you already generate a platform you should copy the boot.scr to image directory and regenerate the platform again to make the mew file work.

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx