Xilinx/Vitis-Tutorials #101: 02-Edge-AI-ZCU104 Vitis Platform Creation Tutorial: Can not run multiple dpu task (multi-thread) in this build.

aclich, 2021-09-03:
I was following https://github.com/Xilinx/Vitis-Tutorials/tree/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104 this platform creation tutorial and refer to this issue https://github.com/Xilinx/Vitis-Tutorials/issues/87 to solve some problem in the flow and build the project successfully. However, I was trying some examples from vitis-ai-library and VART, the system will crash without any information while running the program with multiple dpu task or threads. But the program with only one dpu task or thread can run without problem.

I've tried version 2020.2(Vitis-ai 1.3) and 2021.1(vitis-ai 1.4) in this tutorial and both of them have same problem. But the Vitis-ai official image from [here](https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-zcu104-dpu-v2021.1-v1.4.0.img.gz) won't have this problem.

Is there anything need to setup for running multiple dpu task or I missed some thing? Because the system will not print out any error code or info during the crash, it's hard to know which part is going wrong and giving a screenshot. Sorry for the inconvenience.

Premduth, 2021-09-03:
@aclich This may be better suited for the folks in the Vitis AI tutorials:
https://github.com/Xilinx/Vitis-AI-Tutorials

Can you ask there?

imrickysu, 2021-09-04:
One difference between base platform and VAI pre-built image is that Vitis-AI image optimizes AXI throughput with QoS settings. It's documented in Vitis-AI dpu-trd doc. It may impact Vitis-AI application performance but it should not cause crash. 

The crash may debug needs more knowledge from VART or Vitis-AI application level. Please try to get more help from Xilinx Forums or opening issues in Vitis-AI, Vitis-AI-Tutorials.

aclich, 2021-09-04:
@Premduth @imrickysu  Thanks for the reply and advise on this issue. I will ask about this on there. Many thanks!

closed 2021-09-04 by aclich

aclich, 2021-09-04:
Update
@imrickysu I would like to give a big thanks to you that you reminded about the QoS settings. I went to see VItis-AI dpu-trd  doc [here](https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vitis/README.md) and the section 5.5.3 (known issues) just pointed out the default setting of PMIC (irps5401) on ZCU104 can't afford DPU running on heavy loading. I followed the instruction in the [dpu_sw_optimize.tar.gz](https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/app/dpu_sw_optimize.tar.gz) to copy the driver patch into petalinux project and rebuild. This fix the problem nice and clean.

*note: After copying the driver patch into petalinux project will have error during the kernel config, how I solve it is to 
add a blank text file named _"bsp.cfg"_ inside the path `<plnx_proj>/project-spec/meta-user/recipes-kernel/linux/linux-xlnx/`

imrickysu, 2021-09-07:
That's great, @aclich. Thanks for sharing the solution.