Xilinx/Vitis-Tutorials #102: Error when running the system in tutorial "Versal ACAP AI Engine A to Z Bare-metal Flow"

arthurtsr, 2021-09-06:
Hi,

I'm trying to get through the above tutorial, located here:  https://github.com/Xilinx/Vitis-Tutorials/tree/master/AI_Engine_Development/Feature_Tutorials/01-aie_a_to_z

I am using Vitis and Vivado 2021.1.

I have reached the end of the tutorial (end of part 04-ps_application_creation_run_all.md), everything built well, until the section "Run the system". I click on Run as -> Launch Hardware, and I get this error:

[image]

I have also tried using the SD card but I get the same error. Obviously this is not what is expected, but I don't know how to fix this. Could you please help? I have done this tutorial from the beginning multiple times and always got the same error in the end.

I think there is something wrong with the DDR because when I try "mrd 0x0" (or any address in DDR) in the XSCT Console, I get the following error message :

> "Memory read error at 0x0. Blocked address 0x0. AXI AP transaction error, DAP status 0x30000021"

Thanks,

Premduth, 2021-09-07:
@allyzhou  can you have Joyce look a this?

arthurtsr, 2021-09-17:
Any idea?

paulscooper, 2022-01-14:
I had this same issue. Its resolved now for me by doing the following.....

At the end of 04.Step 2, before you build A-to-Z_app (Hardware) do the following
1) Right Click on A-to-Z_app > Generate Linker Script
2) Change Heap Size from 1k to 1000k
3) Change Stack Size from 1k to 1000k
4) Click Yes when it prompts you to overwrite.
5) Finish Tutorial

> I'm trying to get through the above tutorial, located here: https://github.com/Xilinx/Vitis-Tutorials/tree/master/AI_Engine_Development/Feature_Tutorials/01-aie_a_to_z
>
> I am using Vitis and Vivado [...]

imrickysu, 2022-01-17:
Thanks @paulscooper .

The AtoZ tutorial is also updated with this instruction in https://github.com/Xilinx/Vitis-Tutorials/commit/c1885ba961f07a5c5f4d207bbdc3f565e176e127

The current version should be correct now. https://github.com/Xilinx/Vitis-Tutorials/blob/e4f7ce79838afe3d8884f51e1834ab0460c5a7a6/AI_Engine_Development/Feature_Tutorials/01-aie_a_to_z/04-ps_application_creation_run_all.md

closed 2022-01-17 by imrickysu

shengtaocc, 2025-11-17:
Hello! I would like to ask you a question. Currently, I am using the 2024.2 version of Vitis to implement the Standalone routine. I have implemented the A-to-Z case. When implementing the official routine 04-ps_application_creation_run_all.md, I encountered a cross-compilation error. The header files such as adf.h could not be recognized. I don't know if you can see my message in time. Thank you very much!! This problem has been bothering me for a long time. If you see it, please reply to me. Thank you.