Xilinx/Vitis-Tutorials #108: pseudo won't build in the fasttrack for step2 in the zcu104 tutorial

arrow53, 2021-09-21:
running `make all` from here

https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.md#fast-track

I get this
```
ERROR: Task (virtual:native:/opt/Vitis-Tutorials-2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/ref_files/step2_petalinux/build/petalinux/components/yocto/layers/core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_compile) failed with exit code '1'

```

i'm tried cleaning and rebuilding but that didn't help

imrickysu, 2021-09-21:
Hi @arrow53 , 

This error is in PetaLinux build. It hasn't been reported before. It may relate to your OS environment or network.

There should be some log files in the build directory. Could you take a look the detailed error message?

`/opt/Vitis-Tutorials-2021.1/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/ref_files/step2_petalinux/build/petalinux/components/yocto/layers/core/meta/recipes-devtools/pseudo/pseudo_git*.log`

arrow53, 2021-09-22:
I think this is the issue
```
| /opt/petalinux/2021.1/components/yocto/buildtools_extended/sysroots/x86_64-petalinux-linux/usr/bin/../lib/gcc/x86_64-petalinux-linux/10.2.0/../../../../x86_64-petalinux-linux/bin/ld: cannot find -lgcc_s
```

According to this 

https://www.xilinx.com/support/answers/73705.html

I tried to add but didn't help
```
~/Vitis-Tutorials/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/ref_files/step2_petalinux/build/petalinux/project-spec/meta-user/conf$ cat petalinuxbsp.conf 
#User Configuration

#OE_TERMINAL = "tmux"

IMAGE_INSTALL_append += " libgcc-dev"

```

arrow53, 2021-09-22:
@imrickysu i'm not sure why this is in there but after commenting this line out everything works fine. I'll go ahead and close as I have a fix but feel free to comment if this is really needed
```
  # For GCC < 6.0, enable prebuilt build tools
  #echo "CONFIG_YOCTO_BUILDTOOLS_EXTENDED=y" >> $(PETALINUX_CONFIG)
```

closed 2021-09-22 by arrow53