Xilinx/Vitis-Tutorials #109: exception running on hardware Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104 2021.1

arrow53, 2021-09-22:
any ideas on this? I have no idea where to start
```
root@zcu104_custom_plnx:~# cd /mnt/sd-mmcblk0p1
root@zcu104_custom_plnx:/mnt/sd-mmcblk0p1# ./vadd binary_container_1.xclbin
[   37.920337] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS c
lient for pid(815), ret: 0
[   37.929858] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits
 pid(815)
[   37.939623] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS c
lient for pid(815), ret: 0
[   37.949046] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits
 pid(815)
[   37.957061] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS c
lient for pid(815), ret: 0
INFO: Reading binary_container_1.xclbin
Loading: 'binary_container_1.xclbin'
[   38.987308] [drm] skip kind 29(AIE_RESOURCES) return code: -22
[   38.987319] [drm] found kind 8(IP_LAYOUT)
[   38.993155] [drm] skip kind 9(DEBUG_IP_LAYOUT) return code: -22
[   38.997168] [drm] skip kind 25(AIE_METADATA) return code: -22
[   39.003078] [drm] found kind 7(CONNECTIVITY)
[   39.008816] [drm] found kind 6(MEM_TOPOLOGY)
[   39.013668] cu_drv CU.2.auto:  ffff0000032eb010 xrt_cu_polling_thread: CU[0] 
start
[   39.025509] cu_drv CU.2.auto:  ffff0000032eb010 xrt_cu_polling_thread: CU[0] 
thread end, bad state 0
[   39.034733] [drm] zocl_xclbin_read_axlf 3f3a9d89-c57b-c66d-3457-f0a3f564de8f 
ret: 0
[   39.037830] [drm] bitstream 3f3a9d89-c57b-c66d-3457-f0a3f564de8f locked, ref=
1
[   39.038687] cu_drv CU.2.auto:  ffff0000032eb010 xrt_cu_intr_thread: CU[0] sta
rt
[   39.045508] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_add_context
: Client pid(815) add context CU(0xffffffff) shared(true)
[   39.072087] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_del_context
: Client pid(815) del context CU(0xffffffff)
[   39.083042] [drm] bitstream 3f3a9d89-c57b-c66d-3457-f0a3f564de8f unlocked, re
f=0
[   39.084258] [drm] bitstream 3f3a9d89-c57b-c66d-3457-f0a3f564de8f locked, ref=
1
[   39.091668] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_add_context
: Client pid(815) add context CU(0xffffffff) shared(true)
[   39.114319] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_add_context
: Client pid(815) add context CU(0xffffffff) shared(true)
[   39.126495] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_add_context
: Client pid(815) add context CU(0x0) shared(true)
TEST PASSED
[   39.146175] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_del_context
: Client pid(815) del context CU(0xffffffff)
[   39.158210] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_del_context
: Client pid(815) del context CU(0x0)
[   39.168739] zocl-drm amba_pl@0:zyxclmm_drm:  ffff000001a7f010 kds_del_context
: Client pid(815) del context CU(0xffffffff)
[   39.179707] [drm] bitstream 3f3a9d89-c57b-c66d-3457-f0a3f564de8f unlocked, re
f=0
[   39.205650] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits
 pid(815)
```

imrickysu, 2021-09-23:
XRT driver ZOCL prints mixed with application output.
Your application has an output `TEST PASSED`. It's an expected result.

ZOCL prints are due to the kernel log level being set high. You can see the current log level by running the following command:
```
$ cat /proc/sys/kernel/printk
7       4       1       7
```
The first number is the current log level (has nothing to do with xrt.ini verbosity).
To set it back to the typical value of 4, you can use the following command:
```
$ echo "4" > /proc/sys/kernel/printk
$ cat /proc/sys/kernel/printk
4       4       1       7
```
Now, run the test again and you won't see all the [drm] messages.

arrow53, 2021-09-23:
oh thanks @imrickysu the PASSED was buried in there. sorry about that

closed 2021-09-23 by arrow53