Xilinx/Vitis-Tutorials #111: Hardware_Acceleration Introduction

arrow53, 2021-09-24:
In this tutorial

https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Hardware_Acceleration/Introduction/guided_sw_examples.md

 during `make` I see 
```
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[0]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[0]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[2]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[2]
ERROR: [CFGEN 83-2292] --sp tag applied to an invalid port or argument name: m_axi_gmem
```

I'm using a U50 card. Is this the reason?

randyh62, 2021-09-25:
Yes. You can use the platforminfo command to query the resources for a specific platform or shell, like the following:
platforminfo -p xilinx_u50_gen3x4_xdma_2_202010_1

This shows the U50 has HBM banks, numbered 0 to 32. Therefore you need to specify things a bit differently. 
https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/buildingdevicebinary.html#ejl1524519365386

arrow53, 2021-09-25:
The problem was the `config.mk` file is missing a `u` in the platform name
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Hardware_Acceleration/Introduction/design_source/hw_src/config.mk#L4

What might be good is to instead of saying to run `make` at the bottom of this page
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/Hardware_Acceleration/Introduction/guided_sw_examples.md

to refer to the `design_source` directory, which has a lot of good info

https://github.com/Xilinx/Vitis-Tutorials/tree/2021.1/Hardware_Acceleration/Introduction/design_source

closed 2021-09-25 by arrow53