Xilinx/Vitis-Tutorials #126: ERROR: Host application did not complete - pass/end string not found. Exiting as a failure

zjjzby, 2021-11-10:
When I am running this example,
https://github.com/Xilinx/Vitis-Tutorials/blob/2021.1/AI_Engine_Development/Design_Tutorials/01-aie_lenet_tutorial/

I successfully build the entire design. However, when I was running "make run TARGET=hw_emu" for hardware emulation, I get the following error message:
[image]

Is there any clues about how to resolve this problem? 
My OS is Ubuntu 18.04. The Vitis version is 2021.1

Premduth, 2021-11-10:
@dharapxilinx  can you take a look?

JieGH, 2022-02-18:
I am having the same type of error.

I'm using Vitis 2021.2
I am using version of Vitis-Tutorials is  
https://github.com/Xilinx/Vitis-Tutorials/commit/1f26cb39917da856b13533ffd41934973f20e0ca

JieGH, 2022-02-21:
see Xilinx Form 
https://support.xilinx.com/s/question/0D72E0000072A0wSAE/detail?s1oid=00D2E000000nHq7&s1nid=0DB2E000000XdtH&emkind=chatterCommentNotification&emvtk=r1vZeYMGfuWCXikfx8_dsJnGWDyQbKvctZKiwImxXDo%3D&s1uid=0052E00000N49UN&emtm=1645441005925&fromEmail=1&s1ext=0

This is a standing issue to be solved. 
However, a workaround is to run the program manually

-> go into the package directory and run:
`./launch_hw_emu.sh
`
-> After executing .sh file
```
boot
cd /mnt/sd-mmcblk0p1
export XCL_EMULATION_MODE=hw_emu
export XILINX_XRT=/usr
./lenet_xrt.elf a.xclbin
```

allyzhou, 2022-03-03:
Issue fixed with the commitment https://github.com/Xilinx/Vitis-Tutorials/commit/4bc0a98f5dc78e76f3186bf55a5972beabf451f7

closed 2022-03-03 by allyzhou