Xilinx/Vitis-Tutorials #132: Getting_Started ZCU102 error. cannot boot on ZCU102 with "Kernel panic - not syncing: VFS: Cannot open root device "mmcblk0p2" or unknown-block(179,2): error -6"

sunhao-seu, 2021-11-20:
Dear Authors, I have followed the steps and generated the sd_card folder successfully. However, I cannot boot the Linux system on ZCU102 via sd_card.
This is some error information when booting the ZCU102:
`
[    5.550005] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    5.558618] cfg80211: failed to load regulatory.db
[    5.563609] VFS: Cannot open root device "mmcblk0p2" or unknown-block(179,2): error -6
[    5.571528] Please append a correct "root=" boot option; here are the available partitions:
[    5.579879] 0100           65536 ram0
[    5.579880]  (driver?)
...
[    5.710905] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)
[    5.719329] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.4.0-xilinx-v2020.2 #1
[    5.773873] Kernel Offset: disabled
[    5.784378] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2) ]---
`

**I use Vitis 2020.2 with the following scripts:**
`
source /tools/Xilinx/Vivado/2020.2/settings64.sh
source /opt/xilinx/xrt/setup.sh
unset LD_LIBRARY_PATH
export PLATFORM_REPO_PATHS=/home/***/vitis_2020_2/xilinx_platforms/xilinx_zcu102_base_202020_1
export ROOTFS=/home/***/vitis_2020_2/xilinx-zynqmp-common-v2020.2
export SYSROOT=/home/***/vitis_2020_2_SDK/sysroots/aarch64-xilinx-linux/
source /home/***/vitis_2020_2_SDK/environment-setup-aarch64-xilinx-linux

${CXX} -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I/usr/include/xrt -lOpenCL -lpthread -lrt -lstdc++
v++ -c -t hw --config ../../src/zcu102.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo 
v++ -l -t hw --config ../../src/zcu102.cfg ./vadd.xo -o vadd.xclbin
v++ -p -t hw --config ../../src/zcu102.cfg ./vadd.xclbin --package.out_dir package --package.rootfs ${ROOTFS}/rootfs.ext4 --package.sd_file ${ROOTFS}/Image --package.sd_file xrt.ini --package.sd_file app.exe --package.sd_file vadd.xclbin --package.sd_file run_app.sh
`

@imrickysu @vmayoral @shazahmad @stsoe @Premduth Help~

Premduth, 2021-11-20:
@imrickysu  can you look into this?

sunhao-seu, 2021-11-21:
I find it's my fault. I forget to **write the **"sd_card.img"** to the sd_card.
But After I writing  **"sd_card.img"** to sd_card, new error occurs as follows:

`[  163.388665] FAT-fs (mmcblk0p1): Directory bread(block 2048028) failed`

Then I try Vitis IDE on ubuntu with the **SW acceleration templates/Vector Addition** example and completed my first run of a Vitis accelerated application on the ZCU102 card. 

later, I replace the Vector Addition code with the "vadd" code in this tutorial, it successfully builds the sd_card files but still performs a bad result on ZCU102.  I wonder maybe there are some bugs in the ZCU102 Getting_Started tutorial. Or more details should be added~

Conclusion:
1. I can not obtain the right accelerated result on ZCU102 with the ZCU102 Getting_Started tutorial.
2. I complete my first Vitis accelerated application on ZCU102 with the **SW acceleration templates/Vector Addition example on Vitis IDE on Ubuntu**.  By the way, I failed on Windows.
3. I think the ZCU102 Getting_Started tutorial should be optimized.

imrickysu, 2021-12-10:
Hi @randyh62 , could you look at this Getting started guide feedback? Thanks.

randyh62, 2021-12-10:
Hi sunhao-seu, I will look into the issue to see what might be going on. In the mean time feel free to say how the tutorial should be optimized. 

Thanks,
Randy

dongzhang2021, 2021-12-18:
project-spec/meta-user/system-user.dtsi
&sdhci1 {
      no-1-8-v;
      disable-wp;
};

Candice-Yin, 2022-01-19:
@sunhao-seu Hi, I have the same error with you. Have you solved it?

sunhao-seu, 2022-01-25:
> @sunhao-seu Hi, I have the same error with you. Have you solved it?

This is my solution: https://gitee.com/sunhao2020/vitis_tutorial_zcu102.git

AnusheelXilinx, 2024-12-23:
Closing the thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx