Xilinx/Vitis-Tutorials #137: A question about the example/src/host.cpp and vadd.cpp

PeanutzZ-zZ, 2021-12-03:
Hello,
Recently,i'm learning how to use Xilinx platform and i want to calculate two large number(such as "111111111111111111111111111111111111111111" add "222222222222222222222222222222222222" ) by Xilinx platform. 
Next, i tryed to revise the code based on the 'host.cpp' and 'vadd.cpp'.
Next, **MakeFile**,while starting v++ and making xclbin file ,something error happened.
So, i want to know if i made mistakes in host.cpp or vadd.cpp and how can i put the large num in vadd.cpp . 
:>

The code is as follows
**host.cpp**
```
#define CL_HPP_CL_1_2_DEFAULT_BUILD
#define CL_HPP_TARGET_OPENCL_VERSION 120
#define CL_HPP_MINIMUM_OPENCL_VERSION 120
#define CL_HPP_ENABLE_PROGRAM_CONSTRUCTION_FROM_ARRAY_COMPATIBILITY 1
#define CL_USE_DEPRECATED_OPENCL_1_2_APIS

#define DATA_SIZE 4096

#include <vector>
#include <unistd.h>
#include <iostream>
#include <fstream>
#include <CL/cl2.hpp>
#include "ap_fixed.h"
#include "ap_int.h"
std::vector<cl::Device> get_xilinx_devices();
char *read_binary_file(const std::string &xclbin_file_name, unsigned &nb);

// ------------------------------------------------------------------------------------
// Main program
// ------------------------------------------------------------------------------------
int main(int argc, char **argv)
{
    // ------------------------------------------------------------------------------------
    // Step 1: Initialize the OpenCL environment
    // ------------------------------------------------------------------------------------
    cl_int err;
    std::string binaryFile = (argc != 2) ? "vadd.xclbin" : argv[1];
    unsigned fileBufSize;
    std::vector<cl::Device> devices = get_xilinx_devices();
    devices.resize(1);
    cl::Device device = devices[0];
    cl::Context context(device, NULL, NULL, NULL, &err);
    char *fileBuf = read_binary_file(binaryFile, fileBufSize);
    cl::Program::Binaries bins{{fileBuf, fileBufSize}};
    cl::Program program(context, devices, bins, NULL, &err);
    cl::CommandQueue q(context, device, CL_QUEUE_PROFILING_ENABLE, &err);
    cl::Kernel krnl_vector_add(program, "vadd", &err);

    // ------------------------------------------------------------------------------------
    // Step 2: Create buffers and initialize test values
    // ------------------------------------------------------------------------------------
    // Create the buffers and allocate memory
    cl::Buffer in1_buf(context, CL_MEM_READ_ONLY, sizeof(ap_int<100>) * DATA_SIZE, NULL, &err);
    cl::Buffer in2_buf(context, CL_MEM_READ_ONLY, sizeof(ap_int<100>) * DATA_SIZE, NULL, &err);
    cl::Buffer out_buf(context, CL_MEM_WRITE_ONLY, sizeof(ap_int<100>) * DATA_SIZE, NULL, &err);

    // Map buffers to kernel arguments, thereby assigning them to specific device memory banks
    krnl_vector_add.setArg(0, in1_buf);
    krnl_vector_add.setArg(1, in2_buf);
    krnl_vector_add.setArg(2, out_buf);

    // Map host-side buffer memory to user-space pointers
    ap_int<100> in1 = (ap_int<100>)q.enqueueMapBuffer(in1_buf, CL_TRUE, CL_MAP_WRITE, 0, sizeof(ap_int<100>) * DATA_SIZE);
    ap_int<100> in2 = (ap_int<100>)q.enqueueMapBuffer(in2_buf, CL_TRUE, CL_MAP_WRITE, 0, sizeof(ap_int<100>) * DATA_SIZE);
    ap_int<100> out = (ap_int<100>)q.enqueueMapBuffer(out_buf, CL_TRUE, CL_MAP_WRITE | CL_MAP_READ, 0, sizeof(ap_int<100>) * DATA_SIZE);
    
    ap_int<100> tt1 ("11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111",10);
    ap_int<100> tt2 ("31111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111",10);
  
    in1.set(tt1);
    in2.set(tt2);

    
    // ------------------------------------------------------------------------------------
    // Step 3: Run the kernel
    // ------------------------------------------------------------------------------------
    // Set kernel arguments
    krnl_vector_add.setArg(0, in1_buf);
    krnl_vector_add.setArg(1, in2_buf);
    krnl_vector_add.setArg(2, out_buf);
    krnl_vector_add.setArg(3, DATA_SIZE);

    // Schedule transfer of inputs to device memory, execution of kernel, and transfer of outputs back to host memory
    q.enqueueMigrateMemObjects({in1_buf, in2_buf}, 0 /* 0 means from host*/);
    q.enqueueTask(krnl_vector_add);
    q.enqueueMigrateMemObjects({out_buf}, CL_MIGRATE_MEM_OBJECT_HOST);

    // Wait for all scheduled operations to finish
    q.finish();

    bool match = true;
    std::cout << "==========" << std::endl;
    std::cout << out.to_string() << std::endl;
  
    delete[] fileBuf;

    std::cout << "TEST " << (match ? "PASSED" : "FAILED") << std::endl;
    return (match ? EXIT_SUCCESS : EXIT_FAILURE);
}

// ------------------------------------------------------------------------------------
// Utility functions
// ------------------------------------------------------------------------------------
std::vector<cl::Device> get_xilinx_devices()
{
    size_t i;
    cl_int err;
    std::vector<cl::Platform> platforms;
    err = cl::Platform::get(&platforms);
    cl::Platform platform;
    for (i = 0; i < platforms.size(); i++)
    {
        platform = platforms[i];
        std::string platformName = platform.getInfo<CL_PLATFORM_NAME>(&err);
        if (platformName == "Xilinx")
        {
            std::cout << "INFO: Found Xilinx Platform" << std::endl;
            break;
        }
    }
    if (i == platforms.size())
    {
        std::cout << "ERROR: Failed to find Xilinx platform" << std::endl;
        exit(EXIT_FAILURE);
    }

    //Getting ACCELERATOR Devices and selecting 1st such device
    std::vector<cl::Device> devices;
    err = platform.getDevices(CL_DEVICE_TYPE_ACCELERATOR, &devices);
    return devices;
}

char *read_binary_file(const std::string &xclbin_file_name, unsigned &nb)
{
    if (access(xclbin_file_name.c_str(), R_OK) != 0)
    {
        printf("ERROR: %s xclbin not available please build\n", xclbin_file_name.c_str());
        exit(EXIT_FAILURE);
    }
    //Loading XCL Bin into char buffer
    std::cout << "INFO: Loading '" << xclbin_file_name << "'\n";
    std::ifstream bin_file(xclbin_file_name.c_str(), std::ifstream::binary);
    bin_file.seekg(0, bin_file.end);
    nb = bin_file.tellg();
    bin_file.seekg(0, bin_file.beg);
    char *buf = new char[nb];
    bin_file.read(buf, nb);
    return buf;
}
```

**vadd.cpp**
```
#include "ap_fixed.h"
#include "ap_int.h"
extern "C" {
	void add(
	        ap_int<100> in1, 
	        ap_int<100> in2, 
	        ap_int<100> out, 
	        int size            
	        )
	{
	#pragma HLS INTERFACE m_axi port=in1 bundle=aximm1
	#pragma HLS INTERFACE m_axi port=in2 bundle=aximm2
	#pragma HLS INTERFACE m_axi port=out bundle=aximm1
	ap_int<100> a_int(in1,10); 
	ap_int<100> b_int(in2,10);
	out= a_int+b_int;	
	}
}
```

Premduth, 2021-12-03:
@randyh62  want to take initial look?

randyh62, 2021-12-03:
Sure. I will take a look and reply.

PeanutzZ-zZ, 2021-12-04:
> 当然。我会看看并回复。

thx.
:-)

randyh62, 2021-12-05:
Hello PeanutzZ-zZ,

What error happens when you try to make the xclbin? Can you provide some additional details? What version of the tools are you using? 

Thanks,
Randy

PeanutzZ-zZ, 2021-12-09:
> Hello PeanutzZ-zZ,
> 
> What error happens when you try to make the xclbin? Can you provide some additional details? What version of the tools are you using?
> 
> Thanks, Randy

Hi, i've solved this problem and i can already calculate two bigInteger. However ,when i run app.exe, terminal prints as follow
```
INFO: Found Xilinx Platform 
INFO: Loading 'add.xclbin'
terminate called after throwing an instance of '__gnu_cxx::recursive_init_error'
  what(): std::exception
Segmentation fault (core dumped)
```  

vadd.cpp as follow:
```
#include "ap_fixed.h"
#include "ap_int.h"
extern "C" {
	void add(
	        const ap_int<1024>* in1, 
	        const ap_int<1024>* in2, 
	        ap_int<1024> out, 
	        int size            
	        )
	{
	#pragma HLS INTERFACE ap_none port=in1 bundle=aximm1
	#pragma HLS INTERFACE ap_none port=in2 bundle=aximm2
	#pragma HLS INTERFACE ap_none port=out bundle=aximm1
	out[0]= in1[0]+ in2[0];	
	}
}
```

host.cpp as follow:
```
#define CL_HPP_CL_1_2_DEFAULT_BUILD
#define CL_HPP_TARGET_OPENCL_VERSION 120
#define CL_HPP_MINIMUM_OPENCL_VERSION 120
#define CL_HPP_ENABLE_PROGRAM_CONSTRUCTION_FROM_ARRAY_COMPATIBILITY 1
#define CL_USE_DEPRECATED_OPENCL_1_2_APIS

#define DATA_SIZE 4096

#include <vector>
#include <unistd.h>
#include <iostream>
#include <fstream>
#include <CL/cl2.hpp>
#include "ap_fixed.h"
#include "ap_int.h"
std::vector<cl::Device> get_xilinx_devices();
char *read_binary_file(const std::string &xclbin_file_name, unsigned &nb);

// ------------------------------------------------------------------------------------
// Main program
// ------------------------------------------------------------------------------------
int main(int argc, char **argv)
{
    // ------------------------------------------------------------------------------------
    // Step 1: Initialize the OpenCL environment
    // ------------------------------------------------------------------------------------
    cl_int err;
    std::string binaryFile = (argc != 2) ? "vadd.xclbin" : argv[1];
    unsigned fileBufSize;
    std::vector<cl::Device> devices = get_xilinx_devices();
    devices.resize(1);
    cl::Device device = devices[0];
    cl::Context context(device, NULL, NULL, NULL, &err);
[... 73 lines cut ...]
    }

    std::vector<cl::Device> devices;
    err = platform.getDevices(CL_DEVICE_TYPE_ACCELERATOR, &devices);
    return devices;
}

char *read_binary_file(const std::string &xclbin_file_name, unsigned &nb)
{
    if (access(xclbin_file_name.c_str(), R_OK) != 0)
    {
        printf("ERROR: %s xclbin not available please build\n", xclbin_file_name.c_str());
        exit(EXIT_FAILURE);
    }
    //Loading XCL Bin into char buffer
    std::cout << "INFO: Loading '" << xclbin_file_name << "'\n";
    std::ifstream bin_file(xclbin_file_name.c_str(), std::ifstream::binary);
    bin_file.seekg(0, bin_file.end);
    nb = bin_file.tellg();
    bin_file.seekg(0, bin_file.beg);
    char *buf = new char[nb];
    bin_file.read(buf, nb);
    return buf;
}
```

randyh62, 2021-12-09:
Hello,  This may be related to this post:
https://support.xilinx.com/s/question/0D52E00006ihSKGSA2/vitistutorial-%E8%BF%90%E8%A1%8Cappexe%E6%97%B6%E5%81%B6%E5%8F%91-terminate-called-after-throwing-an-instance-of-gnucxxrecursiveiniterror?language=en_US

I believe it is suggesting changing the -lpthread option to -pthread in the g++ command when compiling the host application. Can you see if that works? 

Thanks,
Randy

PeanutzZ-zZ, 2021-12-13:
> Hello, This may be related to this post: [...]

Hello,thanks for your answer.l've solved this problem and runned the code successfully. 
:P

Thanks,
PeanutzZ-zZ