Xilinx/Vitis-Tutorials #138: When is .run_summary generated

m-kru, 2021-12-04:
[Part5](https://github.com/Xilinx/Vitis-Tutorials/blob/2021.2/Getting_Started/Vitis/Part5.md) mentions `.run_summary` file. However, after successful build of the kernel with Vitis 2020.2 there is no file with `.run_summary` extension.  What program is responsible for generating this file, and when exactly (at which step) this file should be created?

randyh62, 2021-12-10:
The run_summary is generated when the application/xclbin is run and when the opencl_summary and opencl_trace is specified, or other appropriate options are specified, in the xrt.ini file. It is generated when the application is run, not when it is built. The compile_summary and link_summary are generated when the kernels and xclbin are built.

closed 2021-12-10 by randyh62

m-kru, 2021-12-10:
@randyh62 Is `run_summary` generated also when opencl is not used, solely xrt?

randyh62, 2021-12-10:
Yes, but you must have the proper settings in the XRT.INI file. Refer to the following sections of the docs:
https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Enabling-Profiling-in-Your-Application
https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/xrt.ini-File

andy39866821, 2022-04-12:
Hi,
I place _xrt.ini_ in both _hw/_ and _hw_emu/_, but _xrt.run_aummary_ didn't generate.
I doesn't modify source code (host.cpp, vadd.cpp, etc.)
Here are the commands and files I used with U50.
Is there any wrong I made?
```
g++ -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I${XILINX_XRT}/include/ -L${XILINX_XRT}/lib/ -lOpenCL -lpthread -lrt -lstdc++
emconfigutil --platform xilinx_u50_gen3x16_xdma_201920_3 --nd 1
v++ -c -t sw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ../../src/u50.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo 
v++ -l -t sw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ../../src/u50.cfg ./vadd.xo -o vadd.xclbin
```

xrt.ini
```
[Debug]
opencl_summary=true
opencl_trace=true
```

u50.cfg
```
debug=1
save-temps=1

[connectivity]
nk=vadd:1:vadd_1
sp=vadd_1.in1:HBM[1]
sp=vadd_1.in2:HBM[2]
sp=vadd_1.out:HBM[1]

[profile]
data=all:all:all

```

randyh62, 2022-04-13:
Hi andy39866821,

I ran the hw_emu build just to see if it worked. I ran essentially the same commands as you, and used your u50.cfg file and your xrt.ini file contents. It worked fine and generated the xrt.run_summary file as expected. The comparison of your commands and my commends are below, though I did not see an obvious error in your commands. 

**************************************
ANDY:  g++ -Wall -g -std=c++11 ../../src/host.cpp -o app.exe -I${XILINX_XRT}/include/ -L${XILINX_XRT}/lib/ -lOpenCL -lpthread -lrt -lstdc++
RANDY: g++ -Wall -g -std=c++11 /Vitis-Tutorials/Getting_Started/Vitis/example/src/host.cpp -o hw_emu/app.exe \
        -I/opt/xilinx/xrt/include/ -L/opt/xilinx/xrt/lib/ -lOpenCL -pthread -lrt -lstdc++

ANDY:  emconfigutil --platform xilinx_u50_gen3x16_xdma_201920_3 --nd 1
RANDY: emconfigutil --platform xilinx_u50_gen3x16_xdma_201920_3 --od hw_emu --nd 1

ANDY:  v++ -c -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ../../src/u50.cfg -k vadd -I../../src ../../src/vadd.cpp -o vadd.xo 
RANDY: v++ -c -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ./u50.cfg -k vadd -I/Vitis-Tutorials/Getting_Started/Vitis/example/src \
       /Vitis-Tutorials/Getting_Started/Vitis/example/src/vadd.cpp -o hw_emu/vadd.xo

ANDY:  v++ -l -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ../../src/u50.cfg ./vadd.xo -o vadd.xclbin
RANDY: v++ -l -t hw_emu --platform xilinx_u50_gen3x16_xdma_201920_3 --config ./u50.cfg hw_emu/vadd.xo -o hw_emu/vadd.xclbin

ANDY:  setenv XCL_EMULATION_MODE hw_emu 
       ./app.exe
RANDY: cp xrt.ini hw_emu
       cd hw_emu && XCL_EMULATION_MODE=hw_emu ./app.exe

INFO: Found Xilinx Platform
INFO: Loading 'vadd.xclbin'
INFO: [HW-EMU 01] Hardware emulation runs simulation underneath. Using a large data set will result in long simulation times. It is recommended that a small dataset is used for faster execution. The flow uses approximate models for Global memories and interconnect and hence the performance data generated is approximate.
configuring penguin scheduler mode
scheduler config ert(0), dataflow(1), slots(16), cudma(1), cuisr(0), cdma(0), cus(1)
TEST PASSED
INFO::[ Vitis-EM 22 ] [Time elapsed: 0 minute(s) 16 seconds, Emulation time: 0.0784974 ms]
Data transfer between kernel(s) and global memory(s)
vadd_1:m_axi_aximm1-HBM[1]          RD = 16.000 KB              WR = 16.000 KB       
vadd_1:m_axi_aximm2-HBM[2]          RD = 16.000 KB              WR = 0.000 KB        

/Vitis-Tutorials/Getting_Started/Vitis/example/u50> ll hw_emu/
total 63M
-rwxr-xr-x 1 randyh hd 542K Apr 12 20:08 app.exe
-rw-r--r-- 1 randyh hd  769 Apr 12 20:08 emconfig.json
-rw-r--r-- 1 randyh hd  769 Apr 12 20:17 emulation_debug.log
-rw-r--r-- 1 randyh hd 5.8K Apr 12 20:17 opencl_trace.csv
-rw-r--r-- 1 randyh hd 5.2K Apr 12 20:17 summary.csv
-rw-r--r-- 1 randyh hd  62M Apr 12 20:17 vadd.xclbin
-rw-r--r-- 1 randyh hd 9.4K Apr 12 20:17 vadd.xclbin.info
-rw-r--r-- 1 randyh hd  29K Apr 12 20:17 vadd.xclbin.link_summary
-rw-r--r-- 1 randyh hd 197K Apr 12 20:10 vadd.xo
-rw-r--r-- 1 randyh hd 7.2K Apr 12 20:10 vadd.xo.compile_summary
-rw-r--r-- 1 randyh hd  34K Apr 12 20:17 xilinx_u50_gen3x16_xdma_201920_3-0-vadd_simulate.log
-rw-r--r-- 1 randyh hd  944 Apr 12 20:17 xilinx_u50_gen3x16_xdma_201920_3-0-vadd_xsc_report.log
-rw-r--r-- 1 randyh hd   46 Apr 12 20:17 xrt.ini
-rw-r--r-- 1 randyh hd  562 Apr 12 20:17 xrt.run_summary