Xilinx/Vitis-Tutorials #139: How to use the output from script --hls.export_project?

m-kru, 2021-12-04:
v++ has following option:
```
  --hls.export_project arg         Specify a directory where a HLS project
                                   set-up script is exported
```
When I use it, the `.tcl` script is indeed created. However, how can I use this script so that I can open project in `Vitis` or `Vitis HLS` GUI application?

imrickysu, 2022-01-05:
Hi @m-kru , Vitis HLS uses tcl scripts to control the workflow. You can regenerate a Vitis HLS project with commands like `vitis_hls -p xx.tcl`

More info: https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/vitishlscmd.html#wic1584802485878

closed 2022-01-05 by imrickysu

imrickysu, 2022-01-11:
Hi @m-kru , it's the expected behavior. 

Project recreation and project build are separate build targets. User can either append build commands to the HLS project tcl, or use new Makefile targets to build the project. Splitting the actions to finer steps make it easy to control because user can easily add more actions in their workflow but it's not easy to stop in between.

m-kru, 2022-01-11:
@imrickysu if I recreate the project, then build it and then open the Vitis HLS GUI. Should I see the build results?