Xilinx/Vitis-Tutorials #140: unable Export Hardware XSA as Vitis Custom Embedded Platform Creation Example on KV260

dongzhang2021, 2021-12-08:
```
INFO: [Project 1-1042] Successfully generated hpfm file
unmatched open brace in list
ERROR: [Common 17-69] Command failed: unmatched open brace in list
```

imrickysu, 2021-12-10:
Hi @dongzhang2021 , could you attach the logs in Vivado console? I wish to check the Tcl command for export and the full error message. Thanks.

dongzhang2021, 2021-12-10:
hi@imrickysu ,Is this part of the information?Thank you for your help
`start_gui
open_project /media/tnt/Workspace/Xilinx/kv260_202101/zc104_202101.xpr
open_project /media/tnt/Workspace/Xilinx/kv260_202101/zc104_202101.xpr
INFO: [Project 1-313] Project file moved from '/media/tnt/Workspace/Xilinx/zc104_202101' since last save.
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2021.1/data/ip'.
INFO: [IP_Flow 19-3899] Cannot get the environment domain name variable for the component vendor name. Setting the vendor name to 'user.org'.
open_project: Time (s): cpu = 00:00:15 ; elapsed = 00:00:05 . Memory (MB): peak = 7548.859 ; gain = 71.301 ; free physical = 49878 ; free virtual = 57760
update_compile_order -fileset sources_1
set_property pfm_name {xilinx:kv260:nam:0.0} [get_files -all {/media/tnt/Workspace/Xilinx/kv260_202101/zc104_202101.srcs/sources_1/bd/kv260_pl/kv260_pl.bd}]
set_property platform.uses_pr {false} [current_project]
write_hw_platform -hw -force -file /media/tnt/Workspace/Xilinx/kv260_202101/kv260_pl_wrapper.xsa
INFO: [Vivado 12-4895] Creating Hardware Platform: /media/tnt/Workspace/Xilinx/kv260_202101/kv260_pl_wrapper.xsa ...
INFO: [Hsi 55-2053] elapsed time for repository (/tools/Xilinx/Vivado/2021.1/data/embeddedsw) loading 0 seconds
WARNING: [BD 41-2589] Platform should have atleast one axi memory mapped master interface. Enable a master AXI interface as platform AXI_PORT.
INFO: [Project 1-1042] Successfully generated hpfm file
unmatched open brace in list
ERROR: [Common 17-69] Command failed: unmatched open brace in list`

dongzhang2021, 2021-12-10:
``

dongzhang2021, 2021-12-10:
@imrickysu Sorry, I closed the question by mistake

IvoBrandao, 2022-01-25:
I have the same behavior in 2021.2 version
Is there a fix for this issue?

imrickysu, 2022-02-24:
Hi @IvoBrandao , Dongzhang created another thread in https://github.com/Xilinx/Vitis-Tutorials/issues/144 but didn't add more reply. May I know which OS you're running? Is it Windows or Linux? Is there any modifications you made based on the reference files?

IvoBrandao, 2022-02-26:
@imrickysu I'm running Ubuntu 20.04. I tried to create the platform from scratch, so I might have made some mistakes. I just started the Vivado project again and this time everything worked, as the tutorials mention. The weird thing is that I follow the same steps. So not sure why this problem appears. Anyway, thanks for following up on the question :+1:

AnusheelXilinx, 2024-12-23:
Recent issue was addressed in its respective thread. Closing this thread as there are no open concerns.

Thanks
Anusheel

closed 2024-12-23 by AnusheelXilinx