Xilinx/Vitis-Tutorials #141: On (2021.2) Vitis-Tutorials/Hardware_Acceleration/Introduction, Hardware Emulation doesn't work

xooxit, 2021-12-15:
Hi- I'm trying to reproduce Hardware_Acceleration Introduction in various ways.
So I'm gonna build Hardware Design as hardware emulation mode and 
change _Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_src/config.mk_ file like below

```
# Options for TARGET: sw_emu, hw_emu and hw
TARGET ?= hw_emu
# Options for DEVICE: u200, u250. Default platform is XDMA, defined by PLATFORM
DEVICE ?= u250
# If other some specific platforms needs to be used, provide platform path directly
PLATFORM ?= xilinx_$(DEVICE)_xdma_201830_2
# If your platform is not in the standard install area edit this line
PLATFORM_REPO_PATH ?= /opt/xilinx/platforms/
```

After changing _config.mk_ file, I follow **Building the Hardware Design** procedure and **Building the Software Design** procedure. 
But I get a XRT error like below.
```
XRT build version: 2.6.655
Build hash: 2d6bfe4ce91051d4e5b499d38fc493586dd4859a
Build date: 2020-05-22 12:05:03
Git branch: 2020.1
PID: 8513
UID: 1001
[Wed Dec 15 21:09:21 2021]
HOST: user
EXE: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/build/00_load_kernels
[XRT] ERROR: See dmesg log for details. err=-22
[XRT] ERROR: Failed to load xclbin.
terminate called after throwing an instance of 'LineException'
  what():  (xilinx_ocl_helper.cpp:72) Provided XCLBIN is not compatible with any system device
   (Invalid argument)

Aborted (core dumped)
```

The dmesg is as followed. (Being bolded means RED printed dmesg.)

> [  741.004555] xocl 0000:af:00.1: icap.u.16777216 ffff912aaf541010 icap_cached_ocl_frequency: no cached data for 3
> [  741.017809] xocl 0000:af:00.1: icap.u.16777216 ffff912aaf541010 icap_cached_ocl_frequency: no cached data for 3
> [  741.040873] xocl 0000:af:00.1: rom.u.0 ffff911ab72a1810 verify_timestamp: Shell timestamp: 0x5d14fbe6
> [  741.040875] xocl 0000:af:00.1: rom.u.0 ffff911ab72a1810 verify_timestamp: Verify timestamp: 0x5d14fbe6
> [  741.054938] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: trying to find section header for axlf section 20
> [  741.054941] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 4
> [  741.054942] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 9
> [  741.054943] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 0
> [  741.054944] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 6
> [  741.054945] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 8
> [  741.054947] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 7
> [  741.054948] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 14
> [  741.054949] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 2
> [  741.054950] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: saw section header: 22
> [  741.054951] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_axlf_section_header: could not find section header 20
> [  741.054953] [drm] Finding MEM_TOPOLOGY section header
> [  741.054953] [drm] Section MEM_TOPOLOGY details:
> [  741.054954] [drm]   offset = 0x17f1528
> [  741.054955] [drm]   size = 0x148
> [  741.054961] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_cleanup_mem: Taking down DDR : 0
> [  741.054974] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_cleanup_mem: Taking down DDR : 1
> [  741.054984] xocl 0000:af:00.1: icap.u.16777216 ffff912aaf541010 get_axlf_section_hdr: could not find section header 20
> [  741.054988] xocl 0000:af:00.1: rom.u.0 ffff911ab72a1810 verify_timestamp: Shell timestamp: 0x5d14fbe6
> [  741.054989] xocl 0000:af:00.1: rom.u.0 ffff911ab72a1810 verify_timestamp: Verify timestamp: 0x5d14fbe6
> [  741.054992] xocl 0000:af:00.1: icap.u.16777216 ffff912aaf541010 __icap_download_bitstream_axlf: incoming xclbin: 50c278a2-318c-4c17-8eeb-b9898d680e18
>                on device xclbin: 8ce40966-a276-4968-ab5c-8f9d09919e86
> [  741.054995] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev cu, cdev 0000000000000000
> [  741.055179] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev trace_s2mm, cdev 00000000ab9f32a3
> [  741.055353] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev trace_funnel, cdev 00000000fb5c295f
> [  741.055442] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev accel_mon, cdev 000000009e76f7fc
> [  741.055516] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev aximm_mon, cdev 000000007039fb09
> [  741.055586] xocl 0000:af:00.1:  ffff912ab9c140b0 __xocl_subdev_destroy: Destroy subdev aximm_mon, cdev 00000000d0b93658
[... 84 lines cut ...]
> [  741.071131] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Size:0x20000
> [  741.071132] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Type:2
> [  741.071134] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Used:0
> [  741.071135] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Memory Bank: PLRAM[2]
> [  741.071137] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Base Address:0x1000040000
> [  741.071138] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Size:0x20000
> [  741.071139] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Type:2
> [  741.071141] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Used:0
> [  741.071142] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Memory Bank: PLRAM[3]
> [  741.071143] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Base Address:0x1000060000
> [  741.071145] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Size:0x20000
> [  741.071146] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Type:2
> [  741.071147] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   Used:0
> [  741.071149] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: Allocating Memory Bank: bank0
> [  741.071151] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   base_addr:0x0, total size:0x400000000
> [  741.071152] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: Found a new memory region
> [  741.071161] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: drm_mm_init called
> [  741.071162] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: Allocating Memory Bank: bank1
> [  741.071164] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem:   base_addr:0x400000000, total size:0x400000000
> [  741.071165] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: Found a new memory region
> [  741.071173] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_init_mem: drm_mm_init called
> [  741.071176] xocl 0000:af:00.1:  **ffff912ab9c140b0 xocl_read_axlf_helper: Failed to download xclbin, err: -22**
> [  741.266018] [drm] client exits pid(8513)
> [  741.266028] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_drvinst_close: CLOSE 2
> [  741.266033] xocl 0000:af:00.1:  ffff912ab9c140b0 xocl_drvinst_close: NOTIFY 00000000dbef9e1e

randyh62, 2021-12-15:
I am not sure what the specific problem is based on the approach you have taken. However, you can specify the TARGET=hw_emu and the PLATFORM=xilinx_u200_gen3x16_xdma_1_202110_1 in the Makefile command line:

make TARGET=hw_emu PLATFORM=xilinx_u200_gen3x16_xdma_1_202110_1
v++ --kernel vadd --platform xilinx_u200_gen3x16_xdma_1_202110_1 -t hw_emu -s -g -c -o vadd.xo vadd.cpp
v++ --kernel wide_vadd --platform xilinx_u200_gen3x16_xdma_1_202110_1 -t hw_emu -s -g -c -o wide_vadd.xo wide_vadd.cpp
v++ --kernel resize_accel_rgb --platform xilinx_u200_gen3x16_xdma_1_202110_1 -t hw_emu -s -g -I. -I./Vitis_Libraries/vision/L1/include --config vision_config.ini -c -o resize_rgb.xo resize_rgb.cpp
v++ --kernel resize_blur_rgb --platform xilinx_u200_gen3x16_xdma_1_202110_1 -t hw_emu -s -g -I. -I./Vitis_Libraries/vision/L1/include --config vision_config.ini -c -o resize_blur.xo resize_blur.cpp
v++ -l --platform xilinx_u200_gen3x16_xdma_1_202110_1 -t hw_emu -s -g --jobs 47 --config profile.ini --config connectivity_u200.ini -o alveo_examples.xclbin vadd.xo wide_vadd.xo resize_rgb.xo resize_blur.xo --remote_ip_cache ./ip_cache

Can you try this to see if you can run hw_emu?

xooxit, 2021-12-16:
Thanks for reply
 
I have tried to specify TARGET and PLATFORM in the Makefile command line and 
it ends up like below. I think hw_emu runs without errors

```
$ make TARGET=hw_emu PLATFORM=xilinx_u250_xdma_201830_2
v++ --kernel vadd --platform /opt/xilinx/platforms//xilinx_u250_xdma_201830_2/xilinx_u250_xdma_201830_2.xpfm -t hw_emu -s -g -c -o vadd.xo vadd.cpp

......

Successfully wrote (25126053 bytes) to the output file: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/alveo_examples.xclbin
Leaving xclbinutil.
INFO: [v++ 60-1441] [11:11:49] Run run_link: Step xclbinutil: Completed
Time (s): cpu = 00:00:00.14 ; elapsed = 00:00:00.25 . Memory (MB): peak = 1340.742 ; gain = 0.000 ; free physical = 120061 ; free virtual = 128288
INFO: [v++ 60-1443] [11:11:49] Run run_link: Step xclbinutilinfo: Started
INFO: [v++ 60-1453] Command Line: xclbinutil --quiet --force --info /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/alveo_examples.xclbin.info --input /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/alveo_examples.xclbin
INFO: [v++ 60-1454] Run Directory: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/_x/link/run_link
INFO: [v++ 60-1441] [11:11:49] Run run_link: Step xclbinutilinfo: Completed
Time (s): cpu = 00:00:00.41 ; elapsed = 00:00:00.49 . Memory (MB): peak = 1340.742 ; gain = 0.000 ; free physical = 120062 ; free virtual = 128288
INFO: [v++ 60-1443] [11:11:49] Run run_link: Step generate_sc_driver: Started
INFO: [v++ 60-1453] Command Line: 
INFO: [v++ 60-1454] Run Directory: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/_x/link/run_link
INFO: [v++ 60-1441] [11:11:49] Run run_link: Step generate_sc_driver: Completed
Time (s): cpu = 00:00:00 ; elapsed = 00:00:00.01 . Memory (MB): peak = 1340.742 ; gain = 0.000 ; free physical = 120062 ; free virtual = 128289
INFO: [v++ 60-244] Generating system estimate report...
INFO: [v++ 60-1092] Generated system estimate report: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/_x/reports/link/system_estimate_alveo_examples.xtxt
INFO: [v++ 60-586] Created alveo_examples.xclbin
INFO: [v++ 60-1307] Run completed. Additional information can be found in:
        Guidance: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/_x/reports/link/v++_link_alveo_examples_guidance.html
        Steps Log File: /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/_x/logs/link/link.steps.log

INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
    vitis_analyzer /home/lab/yong/fpga-project/tutorial/Vitis-Tutorials/Hardware_Acceleration/Introduction/design_source/hw_emu_src/alveo_examples.xclbin.link_summary 
INFO: [v++ 60-791] Total elapsed time: 0h 3m 37s
INFO: [v++ 60-1653] Closing dispatch client.
```

randyh62, 2021-12-16:
This looks like the build of the xclbin was successful. You still need to run the the application to run the host and xclbin in emulation mode. For example:
```
   setenv XCL_EMULATION_MODE hw_emu
    ./00_load_kernels
```

xooxit, 2021-12-17:
Thanks for reply

NOw works well, it's the issue of unsetting XCL_EMULATION_MODE environment variable and emconfig.json file

closed 2021-12-17 by xooxit